j7126/OctoPrint-Dashboard #266: Themify changes only apply temporarily to tool temperature gauge

Klammerson, 2021-03-19:
All Themify changes apply correctly to the dashboard except for the tool temperature gauge.

CSS changes apply to it but once you open the settings they revert back to standard.

My assumption is that the following part causes the issue:

data-bind="css: $parent.gaugesCentreInGrid('tool', $index())....

While the other elements are like this:

data-bind="css: gaugesCentreInGrid('bed', $index())....

Does this '$parent.' have any reason to be there?

j7126, 2021-03-19:
Hi @Klammerson, yes the $parent is required it will not work without it because it is within a loop. The gaugesCentreInGrid actually has nothing to do with the theme colors, it is for ensuring that the temp gauges are centered. Can you please upload a screenshot of the exact issue you are experiencing.

j7126, 2021-03-19:
@Klammerson I have tried to reproduce the issue that you are having according to your description and have not been able to see the same issue. The only thing that i can think of is if you have the temp gauge colors option enabled which color codes the gauges based on their temperature value.

If this is not the case can you please upload screenshots, your octoprint.log file and if there are any error messages in the browser console.

Klammerson, 2021-03-19:
[image]
[image]
[image]
[image]

The Screenshot from 08:54:14 shows the initial setup. The temperature + fan gauges are aligned left.

The screenshot from 08:55:03 shows the Themify css setting I applied. All dashboardGridItems should align center.

The screenshot from 08:55:18 shows the successful change to the dashboard.

The screenshot from 08:55:35 shows that the tool temperature gauge has moved to the left after I opened the settings and closed them without doing anything else.

No errors in browser console.

The developer tool in safari reveals that the Themify settings aren't applied to the tool gauge anymore but are applied for the other gauges/gridItems.

j7126, 2021-03-19:
I understand and can reproduce the issue now and am working on a fix

j7126, 2021-04-25:
This issue is caused by an underlying issue in Themeify, I can see you have already reported this issue here https://github.com/Birkbjo/OctoPrint-Themeify/issues/88.

I have submitted a pull request to Themeify to fix this issue here https://github.com/Birkbjo/OctoPrint-Themeify/pull/121.

If the developer of Themeify @Birkbjo, accepts my changes and releases it the issue will be fixed.

In the mean time @Klammerson, if you install Themeify directly from my fork via this link `https://github.com/j7126/OctoPrint-Themeify/archive/master.zip` you should be able to see this issue resolved temporarily.

closed 2021-05-18 by j7126