j7126/OctoPrint-Dashboard #267: Slow/laggy octorpint web interface when using 1.18.1

samwiseg0, 2021-03-22:
**Describe the bug**
After upgrading to 1.18.1 and doing any print that takes some time such as an hour or so the webUI is very sluggish and unresponsive. If I revert back to 1.17.0 the issue goes away and the webUI is responsive as it should be. 

I have tried removing all the widgets in the dashboard to see if one in particular was causing the issue. That did not seem to change the responsiveness. I know there was a bug fix with the webcam in the last release but that did not seem to make a difference. 

I checked the network tab in the console as well and nothing seemed out of place. Chrome and safari struggled to even show the developer tools. That seems to indicate its bogging down the browser for some reason. 

**To Reproduce**
Steps to reproduce the behavior:
1. Load octoprint UX
2. Run a print job for some time ~1 hour
3. Observe that the web interface is not very responsive

**Expected behavior**
The webUI should be as responsive as it was with 1.17.0

**Screenshots**

N/A

**Files (please attach the following Files or screenshots when applicable):**
NA

**OctoPrint server and Plugins:**
 - Platform: **Rpi 4B**
 - OS and version: **Raspbian**
 - OctoPrint version: **OctoPrint 1.5.3**
 - OctoPi version: **OctoPi 0.18.0**
 - Python version **Python 3.9.0**
 - Dashboard version **Dashboard 1.18.1**
 - DisplayLayerProgress version **1.25.4**

**Desktop (please complete the following information):**
 - OS and version: **MacOS**
 - **Chrome Version 89.0.4389.82 (Official Build) (x86_64)**, **Safari Version 14.0.3 (16610.4.3.1.7)**

**Logs**
[octoprint (1).log](https://github.com/j7126/OctoPrint-Dashboard/files/6178730/octoprint.1.log)

**Additional context**
NA

Willmac16, 2021-03-22:
Can you elaborate on the procedure to produce this error: Does it happen when you leave the window open for an hour? or would it also happen if you opened up the octoprint web interface an hour into an ongoing print?

Could you also attach a browser console log so we might better track down the issue.

samwiseg0, 2021-03-22:
> Can you elaborate on the procedure to produce this error: Does it happen when you leave the window open for an hour? or would it also happen if you opened up the octoprint web interface an hour into an ongoing print?

This happens when regardless of whether the window is open or not. I have been closing the window because at some point it completely locks up. If I open the page fresh it takes a long time to load and then when it becomes hard to navigate and then locks up completely after 1-2 minutes. 

> Could you also attach a browser console log so we might better track down the issue.

Sure. I can do this tomorrow as I already reverted back to 1.17.0 and I have a print job going. 

I was receiving warnings from the terminal saying that communication was too slow and it was failing back to the basic terminal view. This was understandable as the webUI was unusable. But there were no errors other than that warning in the console.

samwiseg0, 2021-03-22:
After testing a few gcode files that gave me issues yesterday I am unable to reproduce it. I completely wiped the plugin and config and started fresh this morning and that may have "fixed" whatever issue was happening. Thank you for your time in looking into this. 

If it crops back up I will see if I can grab some more information. 

Thanks again

closed 2021-03-22 by samwiseg0