j7126/OctoPrint-Dashboard #268: webcam flip

tutanra, 2021-03-23:
Webcam settings flip do not start when dashboard is active. I have to deactivate it and activate it to have the webcam flip again. This bug did not appear before the last update.

Appears with UI Customizer and DashBoard together.

Regards.

Willmac16, 2021-03-23:
Is this the octoprint control tab's webcam or the dashboard webcam?

tutanra, 2021-03-23:
In control webcam tab the image is correct, webcam is disabled in dashboard settings and /webcam/?action=stream is the problem, doesnt read the init settings, there are a minituare in ui customizer that show flipped also.

When I enable dashboard settings, then, /webcam/?action=strem is ok.

Willmac16, 2021-03-23:
What do you mean by
> webcam is disabled in dashboard settings and /webcam/?action=stream is the problem

tutanra, 2021-03-23:
sorry, when I disable the webcam widget in the dashboard settings, the link /webcam/?action=stream is flipped.

Willmac16, 2021-03-23:
As in, the link url changes to something that does not work or the physical webcam stream is flipped so that it looks backwards?

Thank you for clarifying!

tutanra, 2021-03-23:
My webcam needs to flip and this is active in settings. When I restart octoprint, the webcam is not flipped so I must to disable and enable to show correctly. Seems that webcam doesnt read the settings, only occurs when dashboard & ui customizer and the webcam widget is disabled. Before this update all was working correctly, I have two servers and both occurs the same.

Willmac16, 2021-03-23:
Okay, I'll test it out.

tutanra, 2021-03-23:
Thanks a lot.

Willmac16, 2021-03-23:
I installed UI-Customizer and tested thing out.

I was able to reproduce the issue you have.

This is with the webcam widget disabled:
[image]

and this is with dashboard as a whole disabled:
[image]

I will now try to fix the issue, but I am pretty sure that this is an issue with the UI-Customizer plugin since the control tab webcam view works just fine regardless of the dashboard state.

Willmac16, 2021-03-23:
Okay, I think I figured out the problem. You can try out the fix by installing dashboard with the url `https://github.com/j7126/OctoPrint-Dashboard/archive/refs/heads/development.zip`

tutanra, 2021-03-23:
yes, fix now, thanks.

closed 2021-03-23 by Willmac16

LazeMSS, 2021-04-25:
@tutanra  will this be released soon?

j7126, 2021-04-25:
@LazeMSS, I think you may have meant to @ myself or @Willmac16, @tutanra only reported the issue. 

Anyway I had though that this had been released a while ago but i guess not. I will release it now.