j7126/OctoPrint-Dashboard #271: Possibility to change colors without themiefy

philder, 2021-03-31:
Is it possible to add a feature that allows us to change the colors of gauges and the text without needing themify (which I also think is not working)? As a user of the UI Customizer plugin, I would like to use this dashboard with the colors based on my theme.

Thank you!

Willmac16, 2021-03-31:
It sounds like the best solution here would be to have Dashboard also match ui-customizer color schemes and make sure that css theming plugins properly apply to dashboard if possible.

Willmac16, 2021-05-15:
[image]
I got UICustomizer support working (and cleaned up some minor theming glitches), but I am adding this alongside other changes for 1.19. I will let you know when a working early version of 1.19 is ready to be tested.

Willmac16, 2021-05-15:
@philder I was able to move the code over to the dev branch, so you can install with the url `https://github.com/j7126/OctoPrint-Dashboard/archive/refs/heads/development.zip` and try it out.

j7126, 2021-05-17:
currently as far as i can tell, this method only works with the red night theme

j7126, 2021-05-18:
I have submitted a PR to UI Customizer to add css variable support to a few more themes.

LazeMSS/OctoPrint-UICustomizerThemes#19

j7126, 2021-07-01:
Theming from UI Customizer themes is included as of 1.18.3

It would probably be a good idea to add a setting for custom colors as well at some point to be honest.

closed 2021-10-10 by j7126