j7126/OctoPrint-Dashboard #272: Layer Duration Graph Not Resetting

arimb, 2021-04-19:
**Describe the bug**
The Layer Duration graph doesn't reset after each print. It just continues to add to the x axis starting at zero again.

**Expected behavior**
The graph should reset when a new print is started so layer zero is always on the y axis.

**Screenshots**
[image]
This is from a number of prints in a row. You can see the x axis labels resetting to zero without wiping the old data.

**OctoPrint server and Plugins:**
 - Platform: rpi 3b
 - OctoPrint version: 1.4.2
 - OctoPi version: 0.17.0
 - Python version: ??
 - Dashboard version: 1.18.1
 - DisplayLayerProgress version: 1.26.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version: Chrome 89.0.4389.128

j7126, 2021-04-19:
Hi @arimb, can you please verify what you have currently set in 
Settings > Dashboard > Layer Duration Graph > Visibility > Clear on
Additionally if you could upload your octoprint.log and also check if there are any errors in the browser console it would be helpful.

arimb, 2021-04-19:
Here's what I have for visibility settings:
[image]

No errors in the browser console. Here's my octoprint.log: 
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/6335247/octoprint.log)

arimb, 2021-04-19:
I tried changing the "clear on" setting to "print start" and now it seems to be clearing the graph properly (at the start of each print instead of at the end). I'm fine to leave it that way so you can close the issue if it was just a problem for me.

closed 2021-05-02 by j7126