j7126/OctoPrint-Dashboard #273: Return to Top always returns to top of Dashboard instead of current tab

KenLucke, 2021-05-01:
**Describe the bug**
When clicking on the "Return to Top" arrow (lower right) when scrolled all the way to the bottom of a screen, it always takes you back to the top of the **Dashboard** tab instead of the top of whatever tab you happen to be in.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to a tab other than Dashboard
2. Scroll to bottom
3. Click arrow bottom right ("Top" arrow)
4. Transport yourself back to dashboard tab

**Expected behavior**
Should go to the top of the current tab

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files (please attach the following Files or screenshots when applicable):**

https://user-images.githubusercontent.com/57936294/116793076-6a881500-aa79-11eb-8321-e51c89780f39.mp4

octoprint-systeminfo-20210501123423.zip](https://github.com/j7126/OctoPrint-Dashboard/files/6410172/octoprint-systeminfo-20210501123423.zip)

Willmac16, 2021-05-01:
@KenLucke do you know what plugin is providing the scroll to top button (I don't believe it comes with stock octoprint)?

KenLucke, 2021-05-01:
I thought (ass/u/med) it was Dashboard, which was why I posted the bug here.

j7126, 2021-05-01:
@KenLucke as @Willmac16 said, I can confirm that this return to top button is not part of dashboard and it is probably coming from one of the other plugins you have installed. 

Could you inspect element on the return to top button and send a screenshot of what you see there and we might be able to figure out where it is coming form.

j7126, 2021-05-02:
@KenLucke after further investigation it seems that this return to top button comes from the [Floating Navbar](https://github.com/jneilliii/OctoPrint-FloatingNavbar) plugin. I have notified @jneilliii of this issue.

jneilliii, 2021-05-02:
and a new version was just released to resolve the issue.

KenLucke, 2021-05-02:
You guys are fast. :)

closed 2021-05-02 by j7126