j7126/OctoPrint-Dashboard #274: Settings layout weirdness

scaramonga, 2021-05-12:
**Describe the bug**
Settings all squeezed together as in screenshot

**OctoPrint server and Plugins:**
 - Platform: RPi 4b
 - OS and version: Raspberry OS (latest), Windows 10
 - OctoPrint version: 1.6.1
 - OctoPi version: 0.18
 - Python version 3.9.0
 - Dashboard version 1.18.2
 - DisplayLayerProgress version 1.26.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10 2004
 - Browser and version  Chrome 89.0.4389.90
 - 
[image]

[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/6464228/octoprint.log)

j7126, 2021-05-16:
Hi @scaramonga could you please check for any errors in the browser console.

j7126, 2021-07-01:
Closing due to inactivity

closed 2021-07-01 by j7126