j7126/OctoPrint-Dashboard #278: Dashboard not showing layer progress

Doormat1, 2021-05-23:
Im not sure what the issue is, whether its a slicer issue or a Dashboard issue, but as you can see I dont get any layer progress features on dashboard. 
[image]

Willmac16, 2021-05-23:
@Doormat1 What version of Dashboard is this? And do you have Display Layer Progress installed?

Doormat1, 2021-05-23:
I have display layer progress installed and it is 1.18.2

Willmac16, 2021-05-23:
@Doormat1 Can you share your octoprint.log and a browser log

Doormat1, 2021-05-24:
Here is the octoprint log. Not sure how to get a browser log

Willmac16, 2021-05-24:
Can you upload it on GitHub, since it doesn't seem to have gone through.

Doormat1, 2021-05-24:
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/6534071/octoprint.log
Here it is

Doormat1, 2021-05-25:
I guess Dashboard relies on Display Layer Progress which seems to require Print time genius to work properly, as Ive run M503 on my printer and now Im getting a star next to g code files and Im getting max z height.

Willmac16, 2021-05-25:
Dashboard is moving away from requiring Display layer progress (so the max height would work without either of them), but dashboard will probably continue to benefit from print time genius since gcode simulation of that level is out of scope for dashboard.

Doormat1, 2021-06-02:
I had analysis set to only when Idle, have changed it to always now and PTG seems to be working fine as dose Dashboard now.

closed 2021-06-28 by Willmac16