j7126/OctoPrint-Dashboard #281: Can’t configured dashboard and status bar don’t Show Progress

ItsDominator, 2021-06-06:
Hi all, 

Want to check the raspi temperature today but don’t see it on dashboard. Also the progress in the status bar in top of the side is not shown as before. 

If I deactivate the dashboard plugin, the Progress stays in the status bar. 

I deinstalled the dashboard plugin and install it complete new, same problems. 

System and plugins on newest stable version. 

[image]
[image]
[image]
[image]

Willmac16, 2021-06-06:
@ItsDominator looks like Dashboard is running into and error and won't properly render. Does this error show up on other devices/browsers? If so please send us a browser log so we can address the issue. Otherwise you would have to attempt to get browser logs off of your iOS decice which may take more effort.

ItsDominator, 2021-06-12:
After a reboot I can configure the Dashboard again and get the details as before. So I don’t find a issue jet at the Moment. 
thank you!

closed 2021-06-12 by ItsDominator