j7126/OctoPrint-Dashboard #282: Widget Settings - Instructions

Xiondewah, 2021-06-16:
Could you give me some examples of what goes into the widget settings fields please?  All I seem to get is an empty box that I can't type anything into and I can't see what to use them for.

Thanks!

j7126, 2021-06-16:
I am not exactly sure which box you are referring to. Can you send a screenshot of the issue?

Xiondewah, 2021-06-16:
Sorry - I should have done that!

[image]

j7126, 2021-06-16:
Well it looks like the settings are missing for some reason. Does this happen to the settings for all the widgets or is it just one of them? 

Can you check for any errors in the browser console and also upload your octoprint.log file?

Xiondewah, 2021-06-16:
Yes it happens for every widget.

How do I check in the browser console?  Log attached.

[octoprint-2.log](https://github.com/j7126/OctoPrint-Dashboard/files/6662969/octoprint-2.log)

j7126, 2021-06-16:
You can check the console by right clicking somewhere on the page, then press inspect, then the console tab at the top

Xiondewah, 2021-06-16:
[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (eeprom_marlin.css.map, line 0)
> Selected Element
< <div class="modal-body form-horizontal" data-bind="foreach: settings" style="overflow: auto; max-height: 433px;"></div>

Not sure if that's right or not - but it was an error.

j7126, 2021-06-16:
What browser are you using, does this happen in other browsers?

Xiondewah, 2021-06-16:
I'm using Safari on Mac - only browser I have - but I have also checked it on my iPad and it's the same (though obviously that's also safari.

I do have the UI Customizer plugin installed - don't know if that's anything to do with it as there were a couple of glitches on other plugins at one point.

Xiondewah, 2021-06-16:
I just tried Chrome on my iPad and it’s the same I’m afraid.

jbeahan, 2021-06-28:
I am also missing the widget settings.  I am trying to add a second webcam to the dashboard.  Any other way to do this until the bug is fixed?  Any information I can provide to assist?

jbeahan, 2021-06-29:
I went through and disabled all plugins, enabled one at a time, and determined the conflicting plugin in my case was "GCODE System Commands (1.0.1)" by Shawn Bruce

j7126, 2021-06-29:
@jbeahan that's very helpful. I will look at making a fix sometime later today or tomorrow.

j7126, 2021-07-01:
Can you please check if the settings for the following widgets show up: 

- progress gauges
- layer duration graph
- filament widget

j7126, 2021-07-01:
I have released a fix in 1.18.3

closed 2021-07-01 by j7126

jbeahan, 2021-07-01:
I can confirm this fixed the issue!  Thank you so much for your prompt response!!!!

Xiondewah, 2021-07-02:
Yes - thanks for the fix :-)

Xiondewah, 2021-07-03:
Just updated - all working now.  Thanks very much :-)