j7126/OctoPrint-Dashboard #283: Layer Durations graph does not start from zero for follow-on print jobs

StarNamer, 2021-07-05:
**Describe the bug**
The layer duration graph does not clear and start from zero but simply continues from wherever it was at the end of the last job

**To Reproduce**
1. Print a job.
2. Print a second job.
3. The graph is not zeroed.

**Expected behavior**
When a new job is started, the graph should start at zero.

**Screenshots**
[image]

Layer durations graph is on the 4th job and has not zeroed. The information shows the current print s only on layer 8, but the graph shows up to layer more than 132 for the 1st job, 528 for the 2dn and over 88 for the 3rd.

**OctoPrint server and Plugins:**
 - Platform: RPi 3B+
 - OS and version: Raspbian
 - OctoPrint version: 1.6.1
 - OctoPi version: 0.17.0
 - Python version 3.9.2
 - Dashboard version 1.18.3
 - DisplayLayerProgress version 1.26.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version Chrome 91.0.4472.124

**Logs**
[octoprint.zip](https://github.com/j7126/OctoPrint-Dashboard/files/6766424/octoprint.zip)

Willmac16, 2021-07-06:
In Dashboard 1.18, layer duration data comes from Display Layer Progress, so if the graph is continuing on that is possibly related to DLP. For 1.19 Dashboard, moves to use its own layer backend so the issue may disappear. I'll keep my eye out for it and make sure the graph is properly clearing.

j7126, 2021-07-06:
I think there might be an issue in the clear on print end logic. 

@StarNamer can you please try going to `Settings > Dashboard > Layer Duration Graph > Visibility > Clear on` and setting it to clear on print start.

j7126, 2021-07-06:
I think this is the same issue as  #272, looks like it just never got fixed.

StarNamer, 2021-07-13:
Hi, Sorry for not getting back to you sooner., I only just checked the Clear setting and it was set to Ckear on Print End. I've set it to Clear on start and will let you know. I'm mid-print at the moment with a few hours to go.

StarNamer, 2021-07-25:
Apologies again for the slow response. I've done a few print now with the setting "Clear on Start" and it resolves the issue. Thanks.

closed 2021-11-06 by j7126