j7126/OctoPrint-Dashboard #284: Translation

LeBress, 2021-07-06:
Hello,
When do you manage the language translations. I would like to translate it into French.

Willmac16, 2021-07-06:
@LeBress since we haven't done any translations yet it may take a few days to get things ready for you. Once we do, you would be working with a text document (you could even do it on GitHub) and giving a French translation for each snippet of English.

Willmac16, 2021-07-07:
@LeBress if you wanted to get started you could work on [this file](https://github.com/j7126/OctoPrint-Dashboard/blob/c991260cd8467a4a3270b97bab23bfb6f93dbe54/translations/fr/LC_MESSAGES/messages.po). You will need to select the translations branch at the top left.
[image]

Some of the snippets are not appearing correctly in the UI for me, but I think most of what you would need to work with is there.

Let me know if you have questions.

LeBress, 2021-07-10:
Hello,
Here is my translation is finished, it is not perfect and incomplete for the moment. But I would like to see the result before going any further. Comment made on for the update of your plug-in in French.
@ +

----- Mail original -----

image

Willmac16, 2021-07-11:
@LeBress I can't seem to see any translation. Could you try sending it via GitHub on the web rather than by email?

LeBress, 2021-07-12:
Hello,
Here are the messages.po and messages.mo files translated into French. Can you test them to see the different display issues. Thank you.
@ +

[messages.zip](https://github.com/j7126/OctoPrint-Dashboard/files/6800428/messages.zip)

Willmac16, 2021-07-12:
[image]
[image]
[image]

cmd widget settings needs some fixes probably on my part
[image]

[image]

LeBress, 2021-07-12:
Thanks for the test, but there are display problems in other windows. Like you, I will take screenshots that will be more precise and more explicit about the problems.
@ +

Willmac16, 2021-07-12:
I am getting this strange formatting 
[image]

I think most of the issues will be in the Info sections of individual widget's settings.

Willmac16, 2021-07-14:
@LeBress Check out the latest commit on the translations branch. I think I got it all working, however, I had to use a somewhat hacky workaround. I tweaked a few of the translations to match what the French language pack I installed went with, but I will leave the translating up to you.

For anyone who wants to work in the Jinja files, Babel only seems to pick up English strings from `{{ _('English to translate here') }}` style strings but some finicky strings would only translate with `data-bind="text: gettext('English to translate here')"`. The strange behavior mostly appeared with the long text following Info or Warning. There is a hidden section called `testSection` that has the bracket version of the string while gettext is used inside the settings sub-modal.

LeBress, 2021-07-15:
Hello,

Thank you for all the changes you have made so that the French language is better integrated into your plug-in.
Here are the updated message files.

[messages.zip](https://github.com/j7126/OctoPrint-Dashboard/files/6824564/messages.zip)

If I may, here are screenshots of a few issues still present.

I can't translate because I don't understand this sentence

[image]

[image]

[image]

When the mouse cursor is over the text, this is what is displayed

[image]

[image]

When the mouse cursor is on the value here is what is displayed. This is valid for all icons

Why specify that it is the nozzle and not specify that it is the bed in the other graphic

[image]

Thank you for taking all my comments into account. You did a great job.

@ +

Willmac16, 2021-07-15:
@LeBress "Disable nonce for OctoPrint Webcam". Nonce is a one time use random bit of text stuck onto the end of the webcam url to ensure the image is current and not from a while ago. Since Dashboard has its own option to do multiple cameras this setting refers to the webcam configured in the octoprint webcam and Timelapse tab *not* the multiple webcams configured in Dashboard settings. I'm not sure if there is a good translation for nonce (it's the only term I know of that is used in reference to the string).

Willmac16, 2021-07-15:
@LeBress I'll get working on the not yet translated titles and let you know when I'm ready for you to translate them.

I believe the original rational for the nozzle label was for MultiExtruder setups so it would say `Tool 1` `Tool 2` etc., but it shouldn't be difficult to add the labels in for the other gauges.

Were there any other difficult to translate snippets?

Willmac16, 2021-07-19:
@LeBress I found ~30 description strings that I had missed since they weren't formatted how I expected. They are ready to be translated now.

And I added description text to the other temp gauges. (It may still need some tweaking since the text is a bit doubled up).
[image]

I'll try to deal with the fan "Off" string once I have merged the translation code with the actively developed code (since things have changed for the fan stuff on the backend).