j7126/OctoPrint-Dashboard #295: Feedrate graph stroke too wide

vajonam, 2021-11-03:
**Versions**
OctoPrint 1.7.2 
Python 3.8.12
Dashboard: 1.18.3

**Describe the bug**
Looks like the feedrate dial graph width is buggered up.

**To Reproduce**
Steps to reproduce the behavior:

Enable Filament Settings

**Expected behavior**
Graph should like other graphs.

**Screenshots**
[image]

vajonam, 2021-11-03:
This fixes the problem on Chrome inspector.

[image]

vajonam, 2021-11-03:
https://github.com/j7126/OctoPrint-Dashboard/blob/9b5bff518ab1ff133a5d126574e1457ea9cdd8e3/octoprint_dashboard/templates/dashboard_tab.jinja2#L444

j7126, 2021-11-04:
Looks like this should be fixed in version 1.19.0

vajonam, 2021-11-04:
Software updates isn't showing the update yet. Once it does will test and close this. Thanks.

closed 2021-11-05 by j7126