j7126/OctoPrint-Dashboard #296: No more temperatures values in widgets since 1.19

yannglevarec, 2021-11-04:
"Latest version"

**Describe the bug**
I’ve updated Octoprint Dashboard from 1.18.3 to 1.19 this morning. But since the temperature widgets display 0 degrees, even when printing 

**OctoPrint server and Plugins:**
 - Platform: [e.g. RPi 4B
 - OS and version: Raspberry PI 0.18
 - OctoPrint version: 1.7.2
 - OctoPi version: 0.18
 - Python version 3.7.3
 - Dashboard version 1.19
 - DisplayLayerProgress version (removed)

**Desktop (please complete the following information):**
 - OS and version: MacOS last version
 - Browser and version Safari last version

**Logs**
Please upload your OctoPrint log file 

**Additional context**
Add any other context about the problem here.

mikedmor, 2021-11-04:
Also Experiencing this, 

It looks like temperature readings are showing 0 degrees across my entire octoprint instance.

mikedmor, 2021-11-04:
https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/issues/21

closed 2021-11-05 by j7126