j7126/OctoPrint-Dashboard #297: Layer % Progress Thermometer alternates colors and shows too many digits/weird values

KenLucke, 2021-11-04:
Dashboard v 1.19.0

Been a while since I had to report anything significant here.  I didn't feel that reporting "gauges" spelled as "gagues" in the "other settings" tab was enough to bother with :)

**Describe the bug**
Layer progress thermometer alternates colors each update (background/foreground), and shows values with 3 extra (identical) digits that count up incrementally each update.  i.e., if the actual % is 27%, it might show 27000%, then the next update, if it goes to an actual 30%, it will show 30111%, then if the next update is 35%, it will show 35222%, etc.  In the screencast attached, the actual percentages are 4 & 5, but you will see the last three digits incrementing.

This happened ONLY on layer 1 (it's called layer 1 in in the sidebar, but layer 0 in the file.  When layer 2 started, it reverted to correct.  No alternating colors, no extra digits.

I have to wait severakl days to reproduce, because I have a 6.5 day print just starting on one printer (the one on which this was observed), and another 3 days to finish on a second printer.  3rd printer is down for maintenance/moving it, so I may be able to get it back up sooner, but no guarantees.

**To Reproduce**
Steps to reproduce the behavior:
1 Run a print

[image]

https://user-images.githubusercontent.com/57936294/140394794-695fc4df-0a7d-4380-9cf4-ae95694a8e68.mp4

[octoprint-systeminfo-20211104110256.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7477192/octoprint-systeminfo-20211104110256.zip)

j7126, 2021-11-05:
Can you please specify what slicer and slicer version you are using and send the gcode file that you were printing when this occurred.

KenLucke, 2021-11-05:
Using Cura 4.12 (beta), but I've been using that for some time without this issue, I just upgraded Dashboard through the regular upgrade route (notification, "Update Now" dialogs) before this happened.  I cannot share the file, as it's from a NDA protected proprietary model.  I might be able to just send the first two layers.

Fredyy90, 2021-11-05:
Same issue here, using Cura 4.
[CE3_PETG_glasses_stand.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7487908/CE3_PETG_glasses_stand.zip)
11

KenLucke, 2021-11-05:
> Same issue here, using Cura 4.11 [CE3_PETG_glasses_stand.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7487908/CE3_PETG_glasses_stand.zip)

Oh, good, it's always good to see others have the same issues - not that it's good that they have the issue, but good to know I'm not alone, not crazy, and it's not just my system :) :) :)

Fredyy90, 2021-11-05:
worked fine before the plugin update, seems like it doesn't detect the layer changes anymore, my dashboard shows that i'm still on layer 0 with 61k% when i should be at layer 57 and ~50%.
[image]

ChuckRider, 2021-11-05:
Mine is working fine.  Pi4 with Klipper running on the same device.  I 
left all my other plugins in place.  The only config change I made was 
to turn off the enviornment temp in the command widgets section.

Frederick Behrends wrote on 11/5/2021 12:52 PM:

>
> worked fine before the plugin update, seems like it doesn't detect the 
> layer changes anymore, my dashboard shows that i'm still on layer 0 
> with 61k% when i should be at layer 57 and ~50%.
> image 
> <https://user-images.githubusercontent.com/1572327/140556073-b78661a9-74e0-4118-9b72-5af1ff09e292.png>
>
> or Android 

<https://www.postbox-inc.com>

chefjeremylv, 2021-11-05:
Having a similar issue where the print starts normally, and soon the layers start going up way too fast and and the print appears to speed to 100 percent and octo goes idle but the print is just stopped dead. Ran it a second time and had the exact same result (see attached photo). Those were supposed to be about 5 inches tall.
Sliced using cura 4.11 (same file works when printed from USB) 
Octo 1.7.2
latest Dashboard

Everything updated through the regular update processes
[image]
.

j7126, 2021-11-06:
@chefjeremylv that does not appear to be the same as this issue. Please post in a new thread

j7126, 2021-11-06:
Can you please try re-uploading the gcode files and trying again? It is possible that dashboard just needs to re-analyze the files.

chefjeremylv, 2021-11-06:
Yes, I'll replicate the issue tomorrow and take a screen shots etc and make
a new post.

KenLucke, 2021-11-06:
> @KenLucke @Fredyy90, Can you please try re-uploading the gcode files and trying again? It is possible that dashboard just needs to re-analyze the files.

I can't for a few days.  That printer (and that job) are doing a 6.5 day print right now.

j7126, 2021-11-06:
was the file already on the octoprint server before the update or did you upload it after the update?

KenLucke, 2021-11-06:
After.

j7126, 2021-11-06:
Hmm, interesting. Can you make a backup of your OctoPrint instance (excluding uploaded files and timelapses) and send that?

KenLucke, 2021-11-06:
Again, I'll have to wait until this print finishes, but I can do it then.

j7126, 2021-11-06:
i think you can make backups during a print

KenLucke, 2021-11-06:
> i think you can make backups during a print as far as I know

Possibly, but it takes resources away from the print, and I have to have this done properly first time, on time.

Fredyy90, 2021-11-06:
> was the file already on the octoprint server before the update or did you upload it after the update? @KenLucke

cant really say, i updated via cura and opened octoprint interface, don't remember if i updated before or after.
i downloaded the file form octoprint, renamed, reuploaded and started a new print for Layer 1, everything looks fine.
[image]
even at layer 2 it works now, layer progress is correct and in the bottom left, the layer-"counter" also increases.
[image]

> Hmm, interesting. Can you make a backup of your OctoPrint instance (excluding uploaded files and timelapses) and send that?

here you go, just removed my dropbox api key from the backup

[octoprint-backup-20211106-070656.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7489979/octoprint-backup-20211106-070656.zip)
to get the password, direct message me on Twitter @fredyywhv or Facebook @fredyy90 or send me an email to my inbox frederick.behrends on google-mail(gmail).

Fredyy90, 2021-11-06:
[image]

seems to work fine now, even after multiple layers everything works as expected, only issue i still have is: https://github.com/j7126/OctoPrint-Dashboard/issues/301

j7126, 2021-11-06:
I suspect that re-uploading the file should fix the issue for most people, if it does not, please let me know

closed 2021-11-08 by j7126