j7126/OctoPrint-Dashboard #298: Enclosure monitor remains even when marked disabled in settings

KenLucke, 2021-11-04:
Dashboard v 1.19.0

**Describe the bug**
Despite the "Enclosure Gauges" being disabled in the settings:
[image]
The gauge still shows up in the main display:
[image]

[octoprint-systeminfo-20211104110256.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7477295/octoprint-systeminfo-20211104110256.zip)

KenLucke, 2021-11-04:
I was able to remove it by deleting it in this setting panel:

[image]

j7126, 2021-11-05:
You were referring to this one correct?
[image]
That is from the command widgets, as you said you removed the command widget, however you could have also disabled the command widgets in general.
Command widgets now have the option to appear as a temperature gauge.

closed 2021-11-05 by j7126