j7126/OctoPrint-Dashboard #300: Since update OP 1.7.2 Dashboard 1.19.0 Upload via Cura not stored on external folder

SaKiEQ, 2021-11-05:
When sending a file from Cura, the file appears in Octoprint Dashboard and can be printed, but on second attempt to print or when returning / restarting Octo the file disappears, while all previous files (before 1.7.2) are still correctly listed.

I am using external load folder path to a NAS on the same LAN, mounted on the PI. I can see a file being stored on the NAS but with its file name in 'filename.gcode' instead of all previous files filename.gcode.

As you can see the log contains some error in the last lines, related to Dashboard. While OP has no issues with said path nor permission if Dashboard is disabled.... 

PermissionError: [Errno 13] Permission denied: '/home/pi/NASstorage/Print_Files/Ender3/tmpcahdpeui' -> '/home/pi/NASstorage/Print_Files/Ender3/CE3_Head Gear Clip.gcode'

[octoprint-12.log](https://github.com/j7126/OctoPrint-Dashboard/files/7480193/octoprint-12.log)

j7126, 2021-11-06:
Please verify that the user octoprint is being run by has full permissions to access those files.

appleimperio, 2021-11-06:
I'm having exactly the same issue but from Prusa slicer

SaKiEQ, 2021-11-06:
Well it works when using Octoprint without Dashboard plugin. File uploads from Cura, displays in Octoprint UI, stores on NAS and prints just fine. So I somewhat doubt there are basic user permissions issues, unlike Sudo would be required - which I can't enable for Dashboard, unlike some other plugins. 

The same error persists when a .gcode file is exported and uploaded directly via OctoPrint/Dashboard.

appleimperio, 2021-11-06:
Thanks for that I was waiting for my print to finished to try that test. Good to know.

SaKiEQ, 2021-11-06:
Oh and previous version to 1.19.0 that I had installed had no such issues. Nothing on the Pi nor network has changed, apart from the OctoPrint update to 1.7.2 - where the OP folks says its a plugin error not and OP error.

appleimperio, 2021-11-06:
He release a new version 1.19.1 did you try it?

j7126, 2021-11-06:
> He release a new version 1.19.1 did you try it?

1.19.1 does not fix this

j7126, 2021-11-06:
I am now investigating the problem

j7126, 2021-11-06:
what is your octopi / os version?

appleimperio, 2021-11-06:
latest Octoprint running in Docker  64bit

SaKiEQ, 2021-11-06:
the user privileges of the mount are pi:root, as normal when auto mounted by the system, all files stored previously within the folder have the same pi:root rights (drwxr-xr-x). 

My OP is running direct on a Pi (venv).

SaKiEQ, 2021-11-06:
My system info:

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1.2 Safari/605.1.15
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: false
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1909587968
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 21.1.2
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.7.2
printer.firmware: Marlin Ver 1.0.2
systeminfo.generator: systemapi

SaKiEQ, 2021-11-06:
I tried pretty much every combination like using file names without empty spaces, single words, turning Cura thumbnails off, disabling thumbnail plugin, without changes. 

Of corse running OP in Safe mode makes the problem go away, since it also does not load plugins (such as Dashboard).

My best guess its a parsing error of the file path/name.

Kaicura, 2021-11-06:
There is a Bug in version 1.19. 0  and 1.19.1. no gcode Uploads whit cura or drag and drop chrome.
[image]
Going back to 1.18.3,works perfect .
System Info :
browser.user_agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/95.0.4638.69 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1979641856
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin 2.0.9.1 (Aug 19 2021

j7126, 2021-11-06:
It appears that this is being caused by an "invalid cross-device link" error which is being caused by certain python modules used by dashboard to interact with the filesystem not supporting operations between different file systems.

Kaicura, 2021-11-06:
?
[image]

SaKiEQ, 2021-11-06:
> It appears that this is being caused by an "invalid cross-device link" error which is being caused by certain python modules used by dashboard to interact with the filesystem not supporting operations between two different file systems.

Would make sense in my case, as the storage path leads to QNAP NAS on the LAN, which I know can be fidgety with SMB.

But having said that, it worked in previous version, while the system was already on Python3.7. 

What is Dashboard doing different in regards to storage path, then Octoprint on its own ?

appleimperio, 2021-11-06:
I downgrade the Dashboard plugin version from 1.19.0 to 1.18.3 and is working fine I still have others bugs like the wrong layer number but at least is not deleting the files.

appleimperio, 2021-11-06:
> ?
> [image]

@Kaicura your problem seems to be a bit different since in your logs your having an I/O error and that usually means problems with you storage. In our case we can upload the file but the it gets deleted it seems you can't even get the file uploaded.

Kaicura, 2021-11-06:
Upload works  fine  in v 1.18.3,i downgrade it

j7126, 2021-11-06:
This is a very strange issue as it seems that shutil.move is failing when it tries to open the destination file with a no such file or directory error, even though it should not as I have verified that it is present. 

I have made a fix so that it will work for all files uploaded since the update, however the error is currently still occurring for any files uploaded before the update.

j7126, 2021-11-06:
I have release 1.19.2
This error should no longer happen for any files that you upload after updating to 1.19.2

closed 2021-11-06 by j7126

SaKiEQ, 2021-11-06:
I haven't looked into shutil.move it self yet, but I know on a NAS or SMB mount there can be issues when a function like 'copy' checks if there is enough remaining space available, which works fine on a native system but not so nice on mounts. Not saying this is the case with Dashboard, just something I have seen in past...

But thanks so far :)

j7126, 2021-11-06:
the reason why it is strange is that shutil.move is specifically intended to be able to move between different filesystems

SaKiEQ, 2021-11-06:
yeah well - maybe Forrest Gump had it right - "shit happens" - when it comes to my QNAP NAS and SMB implementation nothing surprises me.... and a NAS sole purpose is to host files, yet its not all sunshine on QNAPS. 

At first I thought it was a combination of issues on my end, but apparently others have the same error, plus you can reproduce it too.`

`is shutil.move native Python3.7 or still legacy ?

j7126, 2021-11-06:
it is a builtin python module
https://docs.python.org/3/library/shutil.html
https://docs.python.org/2/library/shutil.html

SaKiEQ, 2021-11-06:
Cool thanks - will spelunk around with it a bit... seems a quiet versatile utility ! 

I will write a few python scripts, so i can poke the system. Then i can see if the console log on the NAS also reveals something of interest...

j7126, 2021-11-06:
cool, I should clarify that dashboard is not actually directly calling any shutil methods, but instead calling an octoprint method which does, and which _should_ work because it works fine when it is called elsewhere but 🤷

SaKiEQ, 2021-11-06:
Ha.... and I was being told off when I 'blamed' OctoPrint at the start, when I filed the first bug report there. 

Ironic ! 

Maybe the @jneilliii could have another look at it from the OP side.

j7126, 2021-11-06:
I am not saying that it is necessarily the fault of octoprint, it could be something do do with how it is used, but i couldn't figure it out

j7126, 2021-11-06:
OctoPrint/OctoPrint#4305

SaKiEQ, 2021-11-06:
As I was saying at the very top, maybe just a path issue... sometimes a simple trailing '/' in the path is required in one function and in the other it throws an error.

j7126, 2021-11-06:
> As I was saying at the very top, maybe just a path issue... sometimes a simple trailing '/' in the path is required in one function and in the other it throws an error.

I don't think that is the case as I have checked the paths

SaKiEQ, 2021-11-06:
just a thought - hence me saying I will play with singular script later to see if I get similar errors when manipulating files on my mount - without going through OP, then at least I can rule out configuration issues with the venv or user permissions.

But before that I will try some sleep..... I am about 20h past my bed time now :)

j7126, 2021-11-06:
The error occurs at the last line here, and as far as I can tell this is all being used correctly 

```python
    path = self._file_manager.path_on_disk(FileDestinations.LOCAL, payload['path'])
    file_object = octoprint.filemanager.util.DiskFileWrapper(payload['name'], path)
    stream = self.createFilePreProcessor(path, file_object)
    stream.save(path)
```

and `self.createFilePreProcessor` returns an `octoprint.filemanager.util.StreamWrapper` and `self.createFilePreProcessor` does work correctly when called by the `octoprint.filemanager.preprocessor` hook

this only occurs when it is a mounted CIFS or similar filesystem

```
Traceback (most recent call last):
  File "/usr/lib64/python3.10/shutil.py", line 805, in move
    os.rename(src, real_dst)
PermissionError: [Errno 13] Permission denied: '/media/shareTest/tmp9l1nhtx8' -> '/media/shareTest/DONOTPRINT2.aw.gcode'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jeff/octoprint/OctoPrint/src/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/jeff/octoprint/OctoPrint/src/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/jeff/octoprint/OctoPrint-Dashboard/octoprint_dashboard/__init__.py", line 500, in on_event
    self.load_from_meta(payload)
  File "/home/jeff/octoprint/OctoPrint-Dashboard/octoprint_dashboard/__init__.py", line 446, in load_from_meta
    stream.save(path)
  File "/home/jeff/octoprint/OctoPrint/src/octoprint/filemanager/util.py", line 103, in save
    with atomic_write(path, mode="wb") as dest:
  File "/usr/lib64/python3.10/contextlib.py", line 142, in __exit__
    next(self.gen)
  File "/home/jeff/octoprint/OctoPrint/src/octoprint/util/__init__.py", line 1019, in atomic_write
    shutil.move(fd.name, filename)
  File "/usr/lib64/python3.10/shutil.py", line 825, in move
    copy_function(src, real_dst)
  File "/usr/lib64/python3.10/shutil.py", line 434, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib64/python3.10/shutil.py", line 256, in copyfile
    with open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/media/shareTest/DONOTPRINT2.aw.gcode'
```

Relevant parts:
octoprint (filemanager/util.py) atomic_write, which calls `shutil.move`, which tries to open the output file for writing (`with open(dst, 'wb') as fdst`).
I have verified that at the time that this occurs, the relevant path, and the relevant file is present and accessible, (although the file being present shouldn't really matter for writing as it should just create it)

SaKiEQ, 2021-11-06:
'//sakieNAS.local/printNAS /home/pi/NASstorage  cifs  user=octo,uid=pi,password=######,iocharset=utf8' 

Thats my mount and CIFS indeed.

SaKiEQ, 2021-11-06:
are doing this a the pi or root user ?

jneilliii, 2021-11-06:
Yeah, permissions denied error is what I mentioned before in the other ticket, but don't think @SaKiEQ picked up on it. You have to use linux commands after connecting the device to configure permissions to grant the `pi` user writes to write data I think.

SaKiEQ, 2021-11-06:
@jneilliii Sure I picked up on it, but since I did 'chmod -R 777 NASstorage' and 'chown -R pi NASstorage' on the entire file path just for good measure so it could be excluded as an problem, yet the problem still persists....

While Octorprint seems fine writing to the same directory on its own, its weird that a plugin can not.

j7126, 2021-11-06:
> Yeah, permissions denied error is what I mentioned before in the other ticket, but don't think @SaKiEQ picked up on it. You have to use linux commands after connecting the device to configure permissions to grant the `pi` user writes to write data I think.

@jneilliii If you look more closely it is actually not a permission denied error, it is a no such file or directory error.

The permission denied error occurs because shutil.move first tries to use os.rename to to move the file, when that fails due to moving to a different filesystem, it is supposed to do it a different way, however `During handling of the above exception, another exception occurred:` `FileNotFoundError: [Errno 2] No such file or directory: '/media/shareTest/DONOTPRINT2.aw.gcode'`

SaKiEQ, 2021-11-06:
thats where my bad path comment came from, it does not find the file, so either read permissions are screwed up, or the path is not working. 

I have opened up all permissions as far as I can think off on linux level, but then the NAS has its own system which I currently dont know what it is using natively (need to look into that). 

Maybe in the translation between file systems something gets hung up. I can read/write on command line level, move and delete files as Pi user (not root). 

I know i had similar difficulties installing some rsync stuff from OS X to the same NAS and there the file path was an issue, if I remember correctly it required a full path with a different mount point naming then the OS it self was showing, not sure if that makes sense to you.

SaKiEQ, 2021-11-06:
This is the rsync script that copies files from OS X to the same NAS, different share point thought.

"rsync -rlptv   /Users/seuschen/Documents/3D\ Print\ Files\ STL admin@192.168.100.200:/share/sakieServer/3D\ Print/3D\ Print\ Files\ STL"

The struggle was with 192.168.100.200:/share/sakieServer/ where it points rsync to 'share' which is not part of the file system, where '/sakieServer/' is what is used in my Octoprint mount path as '/NASstorage/'. 

Will try now and see if when changing the path name, something different happens.

SaKiEQ, 2021-11-07:
Now past the 1.19.2 update the log is clean, what did you change from your end ? It seems to function in every way. 

I have tried the above file path adjustments, both on the mount (stab) and on the Octoprint path, neither works in terms of being able to mount the folder (unlike required from OS X). 

There are a few things on the NAS side we could try, like oplocks disable/enable.

j7126, 2021-11-07:
There was a bug where it would not analyze when you first upload it, and i have fixed that, so now the problem area only occurs when you try to print files that were already uploaded before updating.

SaKiEQ, 2021-11-07:
Super cool - Thanks ! 

So not my fault then, for configuring something totally wrong... jneilliii made it sure sound like that.

appleimperio, 2021-11-07:
@SaKiEQ Did you try the new version with an older file and it gets deleted? I just want to be sure because if that's the case I will keep version 1.18.3

SaKiEQ, 2021-11-07:
@appleimperio no I did not try, but that is not important to me, as I just send the files again. I have double redundancy, where all files on the CAD machine are auto synched to the NAS up on change into a different folder plus OctoPrint on the different printers does the same... so its literally just drag and drop or re-slicing and sending again. 

Just glad we found the bug.

j7126, 2021-11-07:
I mean the actual bug is still not resolved and honestly idk what is causing it, but it is just a workaround for newly uploaded files. older files from before the update will still experience the error @appleimperio @SaKiEQ

j7126, 2021-11-07:
perhaps someone else from the octoprint community could help figure it out

SaKiEQ, 2021-11-07:
So the same issue appears on your machine too, or is it isolated to few people only ?

j7126, 2021-11-07:
> So the same issue appears on your machine too, or is it isolated to few people only ?

yes, I can reproduce the issue by setting the files directory to a mounted CIFS folder

SaKiEQ, 2021-11-07:
ok good - or not good - but good to know its an issue other then just user error / config error of OctoPrint. 
When I posted it first in OP git I got that undertone and then here again - but not from you ;). 

I think attitude is not a great thing when trying to find bugs, regardless of peoples skill levels. 

I am pretty sure somewhere there is similar bug report from other applications.... and a resolution/explanation to it.

jneilliii, 2021-11-07:
> Super cool - Thanks !
> 
> So not my fault then, for configuring something totally wrong... jneilliii made it sure sound like that.

Sorry if I made it sound that way, that was not my intention. I was just seeing the error in the log and validating permissions were added to the NAS. I've seen people have weird issues like this when they don't.

SaKiEQ, 2021-11-07:
@jneilliii thanks for this reply - really appreciate it ! 

I normally due my due diligence before even trying to post a bug, not to waste other peoples really valuable time. But then there is limitations to my overall experience in the project and what it involves. 

So I primarily doubt my self first - hence its good to be able to burry some doubts every once in a while - while finding a solution (and maybe even contributing in a limited way)

jneilliii, 2021-11-07:
> ok good - or not good - but good to know its an issue other then just user error / config error of OctoPrint. When I posted it first in OP git I got that undertone and then here again - but not from [...]

You'll find I'm not a bad guy...I help tons of people in the community and am actively assisting in Discord, the forums, and issue trackers....so once again I apologize if it came across wrong.

SaKiEQ, 2021-11-07:
@jneilliii I know - I see all the good you do and can only imagine how little time there is outside of it !!! 

Seriously glad we got to this position so far :)

And one day I can maybe be more hands on... While I do develop in other fields, my skills on linux and python are still in its infancy.

SaKiEQ, 2021-11-07:
But overall I think it was worthwhile exercise to rule out user error and config errors. 

At least the problem is isolated and we can give @j7126 Jeffrey some assistance in figuring it out.

SaKiEQ, 2021-11-07:
Hi there, 

just got a weird error (again), that is in its own case unrelated - disconnect during printing. Never happened before and the machine are up 24/7 since 2 years. 

yet the error is the same as we where discussing here, only this time not related to a plugin as far I can tell, but it sure looks its the shutil that threw the error. 

I checked the connections to NAS, to Raspi and all is absolutely fine. Nothing other then printing was going on, on the subnet or Raspi nor NAS. So pretty safe to rule out network / latency issues. 

Just thought I let you know, since it might solidify the view that there might be more wrong around shutil and CIFS format.

2021-11-07 16:05:26,041 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-11-07 16:15:43,471 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-11-07 16:15:53,485 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-11-07 16:16:03,500 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-11-07 16:16:13,513 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-11-07 16:16:23,519 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-11-07 16:16:33,524 - octoprint.util.comm - INFO - No response from printer after 6 consecutive communication timeouts, considering it dead.
2021-11-07 16:16:33,549 - octoprint.plugins.enclosure - INFO - Detected Error or Disconnect in Error will call listeners for shutdown_on_error!
2021-11-07 16:16:33,564 - octoprint.plugins.psucontrol - INFO - Firmware or communication error detected. Turning PSU Off
2021-11-07 16:16:33,609 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline after error"
2021-11-07 16:16:34,027 - octoprint.filemanager.analysis - DEBUG - Resuming analyzer
2021-11-07 16:16:34,032 - octoprint.filemanager.analysis - INFO - Starting analysis of local:LED Globe.gcode
2021-11-07 16:16:34,035 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/OctoPrint/venv/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/NASstorage/Print_Files/Ender3/LED Globe.gcode
2021-11-07 16:16:34,110 - octoprint.util.comm - ERROR - Error while processing hook preheat.
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/comm.py", line 1315, in _getGcodeScript
    retval = hook(self, "gcode", scriptName)
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_preheat/__init__.py", line 379, in get_gcode_script_variables
    variables = self.get_temperatures()
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_preheat/__init__.py", line 190, in get_temperatures
    temperatures = self.read_temperatures_from_file(path_on_disk)
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_preheat/__init__.py", line 98, in read_temperatures_from_file
    file = open(path_on_disk, 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/NASstorage/Print_Files/Ender3/LED Globe.gcode'
2021-11-07 16:16:34,118 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-11-07 16:16:34,241 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-11-07 16:16:34,244 - octoprint.plugins.enclosure - INFO - Detected Error or Disconnect in Disconnected will call listeners for shutdown_on_error!
2021-11-07 16:16:34,344 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-11-07 16:16:34,354 - octoprint.plugins.enclosure - INFO - Detected Error or Disconnect in Disconnected will call listeners for shutdown_on_error!
2021-11-07 16:16:37,388 - octoprint.filemanager.analysis - DEBUG - Got output: ''
2021-11-07 16:16:37,388 - octoprint.filemanager.analysis - ERROR - Analysis for local:LED Globe.gcode ran into error: No analysis result found
2021-11-07 16:16:37,790 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2021-11-07 16:16:38,438 - octoprint.filemanager.storage.timings - DEBUG - Entering octoprint.filemanager.storage._list_folder...
2021-11-07 16:16:39,790 - octoprint.filemanager.storage.timings - DEBUG - octoprint.filemanager.storage._list_folder(<octoprint.filemanager.storage.LocalFileStorage object at 0xb55329d0>,'/home/pi/NASstorage/Print_Files/Ender3',base='',force_refresh=False) took 1351.09ms
2021-11-07 16:20:26,042 - octoprint.server.heartbeat - INFO - Server heartbeat <3

j7126, 2021-11-08:
@SaKiEQ there appears to be multiple different things going on here.

First the printer goes unresponsive

```
2021-11-07 16:16:23,519 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-11-07 16:16:33,524 - octoprint.util.comm - INFO - No response from printer after 6 consecutive communication timeouts, considering it dead.
```

Then OctoPrint disconnects from the printer

Then, as it is no longer printing, OctoPrint tries to resume analysis tasks that were ongoing before the print started (as when a print is ongoing it will pause these tasks to save cpu for the print)

It is then that it tries to resume analysis for the file `'/home/pi/NASstorage/Print_Files/Ender3/LED Globe.gcode'`, and as it tries to open the file it gets a no such file or directory error. 

So it seems that at some point during the print the file disappeared or was lost access to. 

Can you confirm that the file is present? If so then it would confirm that there is something going on with OctoPrint having access issues to CIFS mount points.

SaKiEQ, 2021-11-08:
The file was present at start of the print and when I restarted the print - so that is for sure. 

To why it stopped I cant give you an answer either.  I followed the logic as you did with to how or what happened. 
I found it peculiar that the error given relates to the same problem as previous issues, make me think as you stated that there is larger issue at work. 

Now that could as simple as a latency issue  of the LAN. I am connected over Ethernet on either side of the connection, so no Wifi involved. 

The disconnect would indicate that something was wrong with the network for sure, but when I SSHd from another machine into the Pi and to the NAS I had full access.... Which was like 2min after the printer stopped. Then again in the past 2 years it never did this ! 

I do not believe in coincidences, specially when it comes to logical systems. Yet not sure how to report this to OctoPrint without being rejected for the above reasons. Just thought it might give you more clues to why your code does not work....

[7 more replies]