j7126/OctoPrint-Dashboard #302: 1.19 breaks Layer and Height graphs

vajonam, 2021-11-05:
Since update to 1.19.0. Layer and Height graphs aren't reading any values.

[image]

j7126, 2021-11-05:
Hello, can you please let me know what your slicer and version is, and upload your octoprint.log and the gcode file that you were printing.

vajonam, 2021-11-06:
I am using the SuperSlicer 2.3.57. This was working prior to 1.19.0. 

Attached are slicer and logs. 

[BrushHolder-BinderClips_v4-Corner-Extended_PETG_0.16mm.gcode.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7490687/BrushHolder-BinderClips_v4-Corner-Extended_PETG_0.16mm.gcode.zip)

[octoprint.log.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7490694/octoprint.log.zip)

vajonam, 2021-11-06:
Possibly also coincides with the 1.7.2 octoprint upgrade.

zebons, 2021-11-06:
same issue here with Dashboard plugin version 1.19.0 and Octoprint 1.72
[image]
As you can see the total number layer isn't shown (however it's good for Octoprint on the left of the screen (layer 37/76)

j7126, 2021-11-06:
Please try re-uploading the file now @zebons @vajonam

zebons, 2021-11-06:
Done but always issue. I get now a "NaN layers" as Total layers.
[image]

zebons, 2021-11-06:
OK I do a try. As I was working with subfolders to store my Gcode files for printing, I do the following actions : 
launch a print file from the subfolder and see that dashboard infos are KO for total layers
So I stop the print then move that file to the main folder and launch it from here
And now infos on the dashboard screen are fine !
I think that the issue is new from the octoprint 1.7.2 upgrade, it was working fine before with octoprint 1.7.0(but not really sure of that....)

vajonam, 2021-11-07:
I too am using sub folders. My prints are in a /mypints/foo.gcode file. I will test from root folder.

zebons, 2021-11-07:
As I noticed some other strange behaviours since the 1.7.2 upgrade I open an issue on the Octoprint [https://github.com/OctoPrint/OctoPrint/issues/4309](url)
So wait & see

vajonam, 2021-11-07:
With latest 1.19.3, now layers show some quite invalid data.

This was during bed heat up before the print
[image]

zebons, 2021-11-07:
It's seem that a new version 1.19.3 fix the issue of metadata for files in sub folders.... 
Wait for the upgrade proposal for testing and verify if it correct the issue

vajonam, 2021-11-07:
yeah tried. that. just re-uploaded the file after upgrade to 1.19.3 still shows that data.

zebons, 2021-11-07:
I do a test in safe mode and it seems that the 1.7.2 version of Octoprint isn't the origin of my issues. I launch a file stored in a sub folder and when the print has finished, the status infos are correctly reported (text colored in green and uploaded time not changed). So it's a plugin(s) issue... 
I hope that the new version of dashboard'll correct all my issues

vajonam, 2021-11-07:
re upload files and cleaning up all older files seemed to fix it for me. happy to close this issue.

vajonam, 2021-11-07:
looks like 1.19.3 fixes but new metadata is needed so neeed to re-upload

closed 2021-11-07 by vajonam

j7126, 2021-11-08:
This appears to be similar to #297. In both cases it should be fixed by updating to the latest version and then re-uploading the gcode files.

zebons, 2021-11-08:
I confirm issues resolved with 1.19.3.  Thanks