j7126/OctoPrint-Dashboard #303: Octoprint and Dashboard see different total and current layer numbers

joaquinabian, 2021-11-05:
**Describe the bug**
Octoprint and Dashboard see different total and current layer numbers

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print in OctoPrint and select the Dashboard tab
2. Number of total layers, and current layers are different in Octoprint/State and on Dashboard.

**Expected behavior**
Expect the same values.

**Screenshots**
Screenshot attached.

[image]

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OS and version: Octopi 0.18.0
 - OctoPrint version: Octoprint 1.7.2
 - Python version Python 3.7.3
 - Dashboard version 1.19.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version chrome 77.0.3865
 - Slicer version Cura 4.11

**Logs** 
OctoPrint log file attached 
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/7488103/octoprint.log)

closed 2021-11-06 by j7126

vajonam, 2021-11-07:
Is this fixed? I am running 1.19.3 and still seeing wildly different layer numbers

[image]

vajonam, 2021-11-07:
[image]

vajonam, 2021-11-07:
342 is the correct number of layers according to the slicer.. is this 1.7.2 issue? @j7126

commit 06b549e7 (2021-11-06): fix #303