j7126/OctoPrint-Dashboard #306: ellipsis for long file names

vajonam, 2021-11-06:
is it possible to add ellipsis to long file names. I can provide a PR if you like. 

[image]

Here is a solution you can use
https://stackoverflow.com/questions/831552/ellipsis-in-the-middle-of-a-text-mac-style

This looks like a function that will work. 
```
function smartTrim(string, maxLength) {
    if (!string) return string;
    if (maxLength < 1) return string;
    if (string.length <= maxLength) return string;
    if (maxLength == 1) return string.substring(0,1) + '...';

    var midpoint = Math.ceil(string.length / 2);
    var toremove = string.length - maxLength;
    var lstrip = Math.ceil(toremove/2);
    var rstrip = toremove - lstrip;
    return string.substring(0, midpoint-lstrip) + '...' 
    + string.substring(midpoint+rstrip);
}   
```

j7126, 2021-11-06:
huh, I thought we already had this, will have to check

Fredyy90, 2021-11-07:
I think this should be done via css `text-overflow: ellipsis;` so it dynamically adjusts the length when resized.

vajonam, 2021-11-07:
> I think this should be done via css `text-overflow: ellipsis;` so it dynamically adjusts the length when resized.

But this puts the ellipsis at the end of the text not in the middle. Guess could argue that file name will always end with .gcode so but there might other stuff before that's more identifiable.