j7126/OctoPrint-Dashboard #309: Extruder filament widget not working correctly

georgandersen, 2021-11-07:
**Describe the bug**
Extruder widget not counting
**To Reproduce**
Steps to reproduce the behavior:
1. Widget not counting
2. On refreshing the page gives numbers in 1000 of meters 
3. Tried om two different printers, with two slicers (3ds and superslicer) same result
4. Installed version 18.3 and everything works (installed 19.1 and the problem reoccurs) 

**Screenshots**
[image]

[image]

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
* Browser console log
* gcode file that can be used to reproduce the problem

**OctoPrint server and Plugins:**
 - Platform: [e.g. RPi 3b+, Intel x86]
 - OS and version: [e.g. Raspbian Stretch, Ubuntu 18.10, Windows 10] **or** OctoPi version: [e.g. OctoPi 0.16.0]
 - OctoPrint version: [e.g. OctoPrint 1.3.11]
 - Python version [e.g. Python 3.9.0]
 - Dashboard version [e.g. Dashboard 1.10.1]

**Desktop (please complete the following information):**
Windows 10
Chrome
3dS and superslicer
**Logs**
[octoprint (1).log](https://github.com/j7126/OctoPrint-Dashboard/files/7492046/octoprint.1.log)

**Additional context**
Problem solve for now by going back to ver 18.3 at least for me.

georgandersen, 2021-11-22:
Tried 1.19.5, - same issue as previously reported . 
Gone back to 1.18.3

steadyaero, 2021-12-06:
I will cross post my comment here from a different thread with a similar issue: https://github.com/j7126/OctoPrint-Dashboard/issues/315#issuecomment-987051744

j7126, 2021-12-11:
Hi, sorry I have taken so long to get to this issue, please can you upload the gcode file that you were printing when you experienced the issue.

georgandersen, 2021-12-11:
[image]

[krukke 50mm laag_0.2mm_PLA_ENDER3BLTOUCH_36m.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7697347/krukke.50mm.laag_0.2mm_PLA_ENDER3BLTOUCH_36m.zip)

Troox73, 2021-12-14:
Here another one, hope you need it ;-)
[image]
[sample.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7712771/sample.zip)

colindawson, 2022-01-10:
I'm getting this happening as well.  Every print that I do, doesn't matter if I use Cura or Simplify3D to slide.   The filament amount is always wrong.

[image]

The filament length in the dashboard doesn't seem related to the filament used in the print.

colindawson, 2022-01-10:
I've taken a quick look at this it appears from what I'm seeing is that the problem is in the data being fed to the onDataUpdaterPluginMessage function.  So it looks like the data that's comming in from octoprint itself either corrupted, or not in meters.   Looks to me like this needs to be fixed in the octoprint source, not the dashboard.

georgandersen, 2022-01-11:
> Looks to me like this needs to be fixed in the octoprint source, not the dashboard.

The issue started after version 18.3.
So i am used 18.3, where everything works fine, also after upgrading Octoprint.

j7126, 2022-01-12:
> Looks to me like this needs to be fixed in the octoprint source, not the dashboard.

@colindawson the `onDataUpdaterPluginMessage` is just a way of passing data from the back-end of octoprint to the front-end. This line here: `if (plugin == "dashboard") {` means that all these messages are in fact coming from the back-end of dashboard, not from octoprint, so it is indeed an issue with dashboard.

j7126, 2022-01-12:
This is related to the new gcode processing that was introduced recently in dashboard, as it seems to be having some issues interpreting gcode from any Slic3r++ forks, such as PrusaSlicer, SuperSlicer etc.

j7126, 2022-01-12:
[image]

j7126, 2022-01-12:
Ok so I think that I may have solved this issue.
It seems that the cause was dashboard was not correctly handling the G92 GCODE command.
I will make a new release soon.

closed 2022-01-12 by j7126

j7126, 2022-01-12:
New version released

colindawson, 2022-01-12:
@j7126 awesome.   I'm not fully up to speed with Python, just started doing some MicroPython recently (YAY RP PICO) and was trying to apply what I know from other programming languages.  I'm not familiar with the architecture of Octoprint and the plug in mechanism, hence why I thought I was coming from the OctoPrint source.    Sounds like my pointer helped to narrow down where the problem was a bit.      Hopefully, next time, I'll know to keep digging so that I can find the true source of the problem, and not just stop half way to the root.

Thank you for fixing this.  Glad it sounds like it wasn't too difficult in the end.

commit 872ea8f4 (2022-01-12): fix #309