j7126/OctoPrint-Dashboard #310: Format of average Layer time H:mm:ss

miweimann, 2021-11-07:
since 1.90 the format of average Layer time is no more like h:mm:ss, but is something like seconds? This is not useful, because i have to get an calculator every time, i want to know how long the time is in hours minutes and seconds. 

**Describe the solution you'd like**
Please change the format or give a setting to change it by myself.

KixMan28, 2021-11-17:
I don't think this should be an "enhancement" as this feature was present in the previous versions. I would rather call it a "bug" in the current version, unless this is the way the plugin owner intended to implement this.

miweimann, 2021-11-17:
How can i change the label? i am quite new in this.

KixMan28, 2021-11-17:
@miweimann Why you closed this issue?

j7126, 2021-11-18:
I will look at fixing

KixMan28, 2021-11-18:
@j7126 Thank you for having a look at it.

brunoherrera, 2022-01-19:
still present

j7126, 2022-01-27:
Sorry I took so long to get to this, the fix will be in the next release.

closed 2022-01-27 by j7126