j7126/OctoPrint-Dashboard #311: Layer Progress from SuperSlicer Gcode

bhaal2, 2021-11-08:
When using SuperSlicer the resultant gcode is not compatible with the current version of dashboard (1.19.3), layer progress is not calculated or displayed on the Dashboard.  It just shows 0/0 or 0/1 layers.

SuperSlicer 2.3.57 And the current stable version as well.  And this only started happening with the version of Dashboard that stopped using DisplayLayerProgress plugin.

Regards,
Cameron

vajonam, 2021-11-08:
I am using SuperSlicer and it working okay for me. Try re-uploading the prints, there was issue with how metadata was stored. It wasn't specific to SuperSlicer.

vajonam, 2021-11-08:
[image]

See here everything lines up.

bhaal2, 2021-11-08:
Oh, I saw that mentioned in another bug, will see if I can use that bug to resolve it.  Thank you.

bhaal2, 2021-11-09:
I've tried re-installing to clear the cache etc, and uploading new gcode from SuperSlicer, but it's still not giving me layers, or layer time.

j7126, 2021-11-11:
I will have to take a look at SuperSlicer

StefanCohen, 2021-11-11:
It behaves just like PrusaSlicer/Slic3r under normal circumstances but there is probably a setting somewhere that makes it produce layer markers in a non-standard way. I guess diffing a user-modified  profile that generates wonky g-code against a default profile would go a long way.

j7126, 2022-01-12:
also see #315

j7126, 2022-01-12:
@bhaal2 can you please try with a fresh profile on your slicer (reset ALL settings to default). probably easiest to backup your slicer config files and then delete them to start from scratch. and then let me know if the issue persists.
If the issue stops by doing that, can you please send me your slicer profile file that is having the issue

j7126, 2022-03-10:
Are you still experiencing this issue? I have not been able to reproduce it.

closed 2022-03-10 by j7126