j7126/OctoPrint-Dashboard #313: No Upload to Octoprint from Cura after Update with Dashboard v1.19.3 (1.19.0,1.19.1, 1.19.2)

Kaicura, 2021-11-09:
Self Problem after testing  new Version 1.19.3. Roll always back to V1.18.3.
Cura 4.11.

browser.user_agent: Mozilla/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/95.0.4638.69 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1979641856
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin 2.0.9.1 (Aug 19 2021
[image]
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/7506999/octoprint.log)

yodrako, 2021-11-09:
I'm having the same issue with Prusa Slicer.

boeningc, 2021-11-10:
Same issue just trying to upload direct via OctoPrint web interface.

herrla, 2021-11-11:
I'm having the same issue while trying to upload via Web Interface.

OnScreen Error:
[image]

Octoprint Log:
```
2021-11-11 09:03:04,163 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1552, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/files.py", line 651, in uploadGcodeFile
    display=canonFilename,
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 743, in add_file
    display=display,
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/storage.py", line 909, in add_file
    file_object.save(file_path)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/util.py", line 105, in save
    shutil.copyfileobj(source, dest)
  File "/usr/lib/python2.7/shutil.py", line 63, in copyfileobj
    buf = fsrc.read(length)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/util.py", line 217, in read
    processed_line = self.process_line(line)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 66, in process_line
    if re.match(self.layer_move_pattern, line) is not None:
UnboundLocalError: local variable 'line' referenced before assignment
```

j7126, 2021-11-11:
This is due to dashboard 1.19 no longer supporting python 2. You will need to update to python 3.

StefanCohen, 2021-11-11:
`__plugin_pythoncompat__ = ">=2.7,<4"`

Isn't this setting in __init__.py supposed to force OP to disable the plugin at startup if the req is not met?

j7126, 2021-11-11:
yes, it is supposed to, however it has been forgotten to update it, fixing now

StefanCohen, 2021-11-11:
Cool. I always thought it was weird of OP to keep it in __settings__.py instead of setup.py. It makes it so easy to forget it.

cp2004, 2021-11-11:
@j7126 this really should have been mentioned in the release notes or **somewhere** on the plugin page. As it stands there's still ~35% of installs on Python 2 and we have had a ton of issues recently relating to the dashboard plugins' new release 1.19.x across OctoPrint's issue tracker and the forums as well.

This needs to be communicated clearer. I'm definitely not here to complain about dropping Python 2 - by all means go for it, it will make your life easier, but it **must** be communicated.

In your `setup.py` file, add a `python_requires` field to prevent installs on Python 2 to save you even more complaints, like in this example
https://github.com/cp2004/OctoPrint-WLED/blob/16edd3141a86aba3054d8f7a6787b10b83d062de/setup.py#L70

You should also add a note **very** prominently to the readme on this repository and on plugins.octoprint.org.

Sorry for all the complaining, but we've had a lot of issues with this across the community over the last few days, and nobody knew it was because you decided it didn't support Python 2 anymore.

j7126, 2021-11-12:
> @j7126 this really should have been mentioned in the release notes or **somewhere** on the plugin page. 

@cp2004 

Sorry, I feel I should clarify. I actually didn't realize that python 2 support was broken until a few hours ago when I wrote the above replies, as the changes that broke it were not added by myself, and I haven't tested with python 2. Once I saw that python 2 support was in fact broken, I thought that the quickest and easiest way to resolve it was just to let people know that it is not supported python 2, and I have since updated the `__plugin_pythoncompat__` to require python 3.

> As it stands there's still >37% of installs on Python 2

Wow, I didn't realize that many people were still on python 2!
I might have to look at adding back support for python 2.

j7126, 2021-11-12:
Alright, in version 1.19.5 I have added back support for python 2, everything should work now, please confirm.

Kaicura, 2021-11-12:
Now it works with 1.19.5. :)
[image]

closed 2021-11-12 by Kaicura