j7126/OctoPrint-Dashboard #315: The layer progress percentage and progress bar go haywire

TheLion, 2021-11-12:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
The layer progress percentage and progress bar go haywire.

**To Reproduce**
Steps to reproduce the behavior:
Open Dashboard during print.

**Expected behavior**
I expected the correct percentage and progress bar to show.

**Screenshots**
A screenshot doesn't do it justice. so I made a short video ;-)

https://u.pcloud.link/publink/show?code=XZvHqUXZedGMK8TuD3YL8fY6FQkiPFkeaHDV

**Files (please attach the following Files or screenshots when applicable):**
I'm not able to upload the gcode file, not even when I simplify the name and use .txt as extention.

**OctoPrint server and Plugins:**
 - Platform: Raspberry Pi 4 Model B Rev 1.1
 - OS and version: Raspbian GNU/Linux 10 (buster)
 - OctoPrint version: OctoPrint 1.7.2
 - Python version [e.g. Python 3.9.0]
 - Dashboard version Dashboard 1.19.5

**Desktop (please complete the following information):**
 - Windows 10
 - Browser and version FireFox 94.0.1 and Chrome 95.0.4638.69
 - Slicer version PrusaSlicer 2.3.3+win64 and processed by ArcWelder

**Additional context**
I currently notice this only on one of my printers (running the same version of Dashboard, PrusaSlicer, ArcWelder, Octoprint and Raspbian and I printed the exact same gcode several times (a couple of weeks ago) with older versions of Dashboard and back then there were no issues. Haven't printed this gcode on the other printer yet.

KixMan28, 2021-11-17:
Have a similar issue issue. I am using the latest (1.19.5) version nad the layer numbers and the filament length are wrongly displayed. See attached picture:
[image]

Could you please correct theses?

StarNamer, 2021-11-17:
Also 1.19.5. Looks like it not getting the layer info and the layer percentage calculation is way off...
[image]

Guenni75, 2021-12-04:
I have not one Layer and no height.
[image]

steadyaero, 2021-12-06:
My layers seem correct, but I too am experiencing wonky extruded filament length:
[image]
Octoprint version: 1.7.2
Server OS: Windows 10 21H1
Dashboard version: 1.19.5
Python version: 3.10.0
Slicer: Creality Slicer 4.8.2-177
Printer: Creality Ender 3 V2

j7126, 2021-12-11:
Alright, sorry that I have taken so long to get to this issue. Can you please upload the gcode files that you were printing when you experienced the issue, and I will be able to take a look at it from there.

TheLion, 2021-12-12:
> Alright, sorry that I have taken so long to get to this issue. Can you please upload the gcode files that you were printing when you experienced the issue, and I will be able to take a look at it from there.

Sure. How do you want me to send the gcode to you? I wasn't able to upload it in this issue.

j7126, 2021-12-12:
> Sure. How do you want me to send the gcode to you? I wasn't able to upload it in this issue.

Hmm, you usually should be able to upload it here. Perhaps put it in a Pastebin?

TheLion, 2021-12-13:
> > Sure. How do you want me to send the gcode to you? I wasn't able to upload it in this issue.
> 
> Hmm, you usually should be able to upload it here. Perhaps put it in a Pastebin?

Upload doesn't work unfortunately. Maybe it is the size, it is 27Mb.

You can find the gcode file here: https://u.pcloud.link/publink/show?code=XZhCnNXZv3ir352ienjY7kLOeYXDYuvDn0cV

TheLion, 2022-01-08:
I just experienced the same issue. With a much smaller gcode file. 
If you want, I can share it.

j7126, 2022-01-12:
So regarding the extruded filament length issue which was mentioned in this thread, I have created a fix which will be released soon, for more info see #309

j7126, 2022-01-12:
also see #311

j7126, 2022-01-12:
Closing as duplicate of #311

closed 2022-01-12 by j7126

j7126, 2022-01-12:
@TheLion can you please try with a fresh profile on your slicer (reset ALL settings to default). probably easiest to backup your slicer config files and then delete them to start from scratch. and then let me know if the issue persists.

TheLion, 2022-01-14:
> @TheLion can you please try with a fresh profile on your slicer (reset ALL settings to default). probably easiest to backup your slicer config files and then delete them to start from scratch. and then let me know if the issue persists. If the issue stops by doing that, can you please send me your slicer profile file that is having the issue

Sure, will do that and let you know what happened.