j7126/OctoPrint-Dashboard #316: Dashboard is Causing GCode Scripts for Start/Stop/Cancel Not to Run

condraka, 2021-11-16:
Running Current Versions

Octoprint 1.7.2
Dashboard 1.19.5
OctoPi 0.18.0
Python 3.7.3
on a Rasp Pi 3B+ Rev 1.3

GCode Scripts are partially working with Dashboard Plugin Active.  Just updated this weekend to 1.19.5 Dashboard and it stopped working.  Program Related Scripts are not being run.  Non-Program Related Scripts are working.

WORKING:  Serial Connection  Established and Closed 

NOT WORKING:  
Before Print Job Starts
After Print Job Completes
After Print Job Canceled
[octoprint-systeminfo-20211115132259.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7542715/octoprint-systeminfo-20211115132259.zip)

j7126, 2022-02-02:
Hi, believe I have created a fix, please install the development branch by installing the plugin with the below url and let me know if the issue is fixed. 

`https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

condraka, 2022-02-07:
> I’ll give it a whirl this weekend.  I love dashboard and it has sucked turning it off to run…. Thanks for the update.
>
> Carl E. Ondraka
> Sent from my iPhone

Scripts are running as expected.  Fix has corrected script issues.  Thanks

Carl E. Ondraka

alaorath, 2022-03-06:
Has this change been merged to master?  I ran into the same issue.  octoprint.log filled with exceptions ("Error while processing hook dashboard for phase sending and command ") and no completed movement.

Even the default octoprint movement buttons did nothing but generate log errors:

> 2022-03-05 17:10:00,385 - octoprint.util.comm - ERROR - Error while processing hook dashboard for phase sending and command G0 Z1 F200:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4611, in _process_command_phase
    tags=tags,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py", line 797, in process_gcode
    current_layer_progress = int((self.layer_moves / self.layer_move_array[self.current_layer]) * 100)
IndexError: list index out of range

Solution was to reboot the PI, after reboot everything seemed normal again.

Octoprint: 1.7.3
Python: 3.7.3
OctoPi: 0.17.0
Dashboard: 1.19.8

j7126, 2022-03-10:
> Has this change been merged to master?

@alaorath It has not yet, but will be included with the next release, coming soon.

closed 2022-03-10 by j7126