j7126/OctoPrint-Dashboard #319: Layercount doubled using CANVAS (canvad3d.io) slicer from Mosaic

unclej84, 2021-11-25:
**Describe the bug**
Apparently CANVAS writes 2 triggering layer indicators per layer. So the dashboard shows doubled layer numbers for current and total layer as well as strange average layer durations (both entries are directly after each other so that one layer is the real layer time and the other one is nearly 0).

These trigger for Simplify3d and the generic slicer [dict](https://github.com/j7126/OctoPrint-Dashboard/blob/146179ab96ddc7b6cfbf619f058a2c779613e1f6/octoprint_dashboard/__init__.py#L555).

The relevant gcode entries are:

- e.g. `; BEGIN_LAYER_OBJECT z=0.2000 z_thickness=0.2000` (line 199 in sample)
- e.g. `; layer 1, Z = 0.2` (line 212 in sample)

The later one is custom gcode for layer change and could be removed in the setting but is IMO the more common one to find and also used in other plugins.

I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Expected behavior**
Only one dictionary should trigger for the whole file. Maybe it would be possible to check the file first for the best matching dict (e.g. with `if (len([*re.finditer(pattern, text)]) > 0)`) and then just use this dict for the whole file instead of [running thru all dicts on every line to find the first match](https://github.com/j7126/OctoPrint-Dashboard/blob/146179ab96ddc7b6cfbf619f058a2c779613e1f6/octoprint_dashboard/__init__.py#L75).

**Screenshots**
[image]

Sample:
[000014_Ender-3 Pro_Thread_Z-Axis-Knob.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7604731/000014_Ender-3.Pro_Thread_Z-Axis-Knob.zip)

j7126, 2022-03-10:
I have made it store the beginning of the first layer indicator that is matched, and only match future layer indicators that start with that. Hopefully this should fix the issue.

This change will be included in the next release, you can try it out by installing the version from development branch `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

closed 2022-03-10 by j7126

mpaw, 2022-11-11:
@j7126 Is this fix going to be released soon?