j7126/OctoPrint-Dashboard #322: Feedrate / 5 second average value swapped

staffert, 2022-01-02:
**Describe the bug**
It seems to me  like the instantaneous feed-rate needle indicator is actually reporting the 5 second rolling average value, and the average value arc is indicating the instantaneous value...

**To Reproduce**
Steps to reproduce the behavior:
1. Install Dashboard Plugin
2. Set display options to include Feedrate display graph
3. Print any job
4. See behavior above

**Expected behavior**
Expected to see needle indicator as instantaneous value, as indicated in help screen for configuration.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Files (please attach the following Files or screenshots when applicable):**
* use any and all g-code that I have printed with. This is a long standing issue.

**OctoPrint server and Plugins:**
  - OctoPrint version: 1.7.2
[octoprint-systeminfo-20220102154447.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7799769/octoprint-systeminfo-20220102154447.zip)

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version: FireFox 95.0.2 64 bit
 - Slicer version: multiple, typically PrusSlicer 2.3.3 or SuperSlicer 2.3.57.8 

**Logs**
see attached ZIP

j7126, 2022-01-12:
I think that the case is that they are not switched around, but in fact the average 5 seconds value is just buggy and incorrect and jumps around.

j7126, 2022-01-12:
I have made some improvements, it should work a bit better now.

staffert, 2022-01-13:
I've done some testing with the newly released version, but now it seems to drop to a zero value more often.....as an average value, that shouldn't be the case, especially when I am printing large filled sections with virtually no change in feedrate for minutes.
I think there is still something wrong with this functionality.

j7126, 2022-01-13:
Hmm, can you send over your guide file?