j7126/OctoPrint-Dashboard #323: Tab not selected on Octoprint-Startup (always enters temperature tab)

Rene1695, 2022-01-07:
Dear all,

i installed the dashboard plugin on my octoprint server. After a reboot, i figured out, that the dashboard tab is on the most left side which shall force octoprint to enter this tab on startup. 

Unfortunately, everytime i connect to my octoprint server, it will always start on the default "temperature" tab from octoprint, which is right next the dashboard tab. Do i have to set a startup tab in the settings? I didnt found any options. 

Additional Informations:
Raspberry Pi Zero V2 running octoprint v 1.7.2
Dashboard Plugin v.1.19.5

[image]

j7126, 2022-01-12:
First thing to try, if you are accessing octoprint from a bookmark, does that bookmark by any chance have `#temp` at the end?
If that is the case then all you would have to do is remove the `#temp` from the bookmark.
If that is not the case, can you please share your octoprint system info bundle

Rene1695, 2022-01-12:
Removing #temp worked for me, thank you very much! Issue can be closed :)

closed 2022-01-12 by Rene1695