j7126/OctoPrint-Dashboard #326: dashboard constantly has 'busy' rotating animation

r0dnay, 2022-01-18:
The problem

dashboard constantly has 'busy' rotating animation
Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)
If you could not test in safe mode, please state why

currently in print and need some plugins intact
Version of OctoPrint

1.7.2
Operating system running OctoPrint

W10 21H1 19043.1466
Printer model & used firmware incl. version

MARLIN 2.0.9.3
Browser and version of browser, operating system running browser

9.5.0.2 FIREFOX

DASHBOARD V1.19.6

j7126, 2022-01-21:
I assume that you mean the spinner where the webcam goes in dashboard? Can you see your webcam in the control tab. Can you please upload your octoprint system info bundle

j7126, 2022-05-27:
@r0dnay do you still have this issue? If so can you confirm that you are talking about the loading animation at the bottom of the dashboard where the webcam would go or is it something else? Can you confirm that you can see the webcam in the control tab? Please upload the octoprint system info bundle and a screenshot of the dashboard webcam settings.

BaileyMillerSSI, 2022-06-21:
@r0dnay I had a similar problem after updating to the latest OctoPrint version. I checked into the camera settings in the UI and a . had been added before the camera path. I removed that, saved, and reloaded page and now the camera is back to working fine.

closed 2022-10-30 by j7126