j7126/OctoPrint-Dashboard #330: Widget for chamber temp is shown although  printer profile does not have chamber temp

gituemmel01, 2022-01-24:
In the dashboard the widget for chamber temp is always displayed, but in the printer profile (Ender 5) the checkbox for heated chamber temp is not selected. So the widgets shows "NaN" and shoud not be visible.
Plattform Raspberry Pi 4, OctoPrint-Dashboard-1.19.6, OctoPrint 1.7.3

[image]

[image]

ackermannsven, 2022-01-27:
Also, when used with a independent dual extruder IDEX setup, the fan speed and chamber temp overlap:

Raspberry Pi 3B, OctoPrint Dashboard 1.19.7, OctoPrint 1.7.3

j7126, 2022-01-27:
That is in fact a command widget, not the heated chamber widget, as the default "example" command widget is a simulated chamber widget. That should not be visible by default, and it seems that there is something broken with hiding disabled command widgets, which is also causing the overlap issue.

I am working on a fix, but in the meantime a temporary workaround you can use would be to just completely disable command widgets from within dashboard settings.
[image]

ackermannsven, 2022-01-27:
Thanks a lot, this worked for me. I did also try to remove the "simulated chamber", seems also to be fine that way.

closed 2022-01-27 by j7126