j7126/OctoPrint-Dashboard #331: [Bug]: Z-Axis and extruder related values do not populate

aseams, 2022-01-28:
### The problem

When looking at the dashboard in OctoPi, the following values do not change and/or appear at all.
- extruded filament
- next filament change
- estimated total print time
- current layer
- current height
- layer time

I have a BLTouch installed as well as the BLTouch plugin for OctoPi.

### Version of OctoPrint

1.7.3

### Platform

Raspberry Pi 3B+

### Operating system running OctoPrint

OctoPi 0.18.0

### Dashboard version

1.19.8

### Printer model & used firmware incl. version

Ender 3 Pro running Marlin 1.1.2

### Browser and version of browser, operating system running browser

Brave browser on Win 10 20H2

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[SysInfo Bundle](https://github.com/j7126/OctoPrint-Dashboard/files/7962125/octoprint-systeminfo-20220128222023.zip)

[image]

Willmac16, 2022-01-28:
@aseams Can you share a gcode file that produces the behavior?

aseams, 2022-01-28:
Wouldn't let me upload .gcode so had to zip it.
[Guide_Open.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7962240/Guide_Open.zip)

aseams, 2022-01-28:
Just started another print that I uploaded through OctoPi and it shows the values in question. The print which produced this error was loaded directly to the SD card.

Willmac16, 2022-01-28:
That would do it. The way Dashboard does those stats requires the file to be on Octoprint. The formatting of those stats when no data is available could be improved though.

StefanCohen, 2022-01-28:
Wouldn't it be better if we remove the widgets that can't be populated when printing from SD?

j7126, 2022-01-29:
> Wouldn't it be better if we remove the widgets that can't be populated when printing from SD? Or maybe gray them out or display an N/A message?

That would probably be a good idea