j7126/OctoPrint-Dashboard #332: [Bug]:

shakeandbake017, 2022-01-31:
### The problem

1. I have a AM2302 sensor that detects temp and humidity.  On the enclosure tab, it shows correctly with the Fahrenheit symbol.  On the dashboard page, it shows with the correct temp number but with Celsius instead of Fahrenheit.  Please see attached screenshot.  

### Version of OctoPrint

1.7.3

### Platform

Raspberry Pi 3b

### Operating system running OctoPrint

OctoPi 0.18.0

### Dashboard version

1.19.8

### Python version

3.7.3

### Printer model & used firmware incl. version

Ender 3 Max 

### Browser and version of browser, operating system running browser

Microsoft Edge

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[image]
[octoprint-systeminfo-20220131175010.zip](https://github.com/j7126/OctoPrint-Dashboard/files/7974743/octoprint-systeminfo-20220131175010.zip)

shakeandbake017, 2022-02-01:
I see an old post staying it's hard coded to celsius so this can close

j7126, 2022-02-01:
This is a bug

closed 2022-03-10 by j7126