j7126/OctoPrint-Dashboard #336: [Bug]: Filamant used displaying incorrect value.

colindawson, 2022-02-14:
### The problem

I'm currently printing out part of a Hero Me Gen 6.   The print is supposed to use about 8.34 Meters of filament according to Octoprint.    However, having printed 76% of the print, the dash board is saying that I've used about 0.274m.

[image]

### Version of OctoPrint

1.7.3

### Platform

Raspberry PI 4 (2GB)

### Operating system running OctoPrint

OctoPi

### Dashboard version

1.19.8

### Python version

3

### Printer model & used firmware incl. version

Ender 3 Pro  Dual extruder mutating  Marlin 2.0.9.3

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

Here's the g-code that I'm running.  It was created with Simplify3D 4.1.2
[HMG6 Base CR10S-Pro.gcode.txt](https://github.com/j7126/OctoPrint-Dashboard/files/8062326/HMG6.Base.CR10S-Pro.gcode.txt)
[octoprint-systeminfo-20220214161856.zip](https://github.com/j7126/OctoPrint-Dashboard/files/8062336/octoprint-systeminfo-20220214161856.zip)

colindawson, 2022-02-15:
I can confirm that the amount of filament used never changed.   This is something that I have noticed with other prints as well.

j7126, 2022-04-21:
I have made a fix.

This change will be included in the next release, you can try it out by installing the version from development branch using this url `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

closed 2022-04-21 by j7126