j7126/OctoPrint-Dashboard #338: Layer Percent Displays Crazy Values

mmikem, 2022-03-25:
Layer Percent is currently displaying 2027645% of current layer complete.  I attached a video..

https://user-images.githubusercontent.com/10793065/160046603-6f01d584-a7c0-4554-ad93-b568027eba2a.mp4

.

mmikem, 2022-04-20:
NOTHING?

j7126, 2022-04-20:
I have not yet had time to take a look at this. 

Of course I will not be able to debug the issue unless you provide some more information. Please upload your octoprint [system info bundle](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) and the gcode file you were printing when the issue occurred. 

I have seen someone reporting an issue like this before but if i remember correctly I could not figure out the cause as they did not provide any other information other than the screenshot.

mmikem, 2022-04-23:
Hi, System info bundle attached. Thanks

[octoprint-systeminfo-20220423010311.zip](https://github.com/j7126/OctoPrint-Dashboard/files/8545215/octoprint-systeminfo-20220423010311.zip)
.

OscarVsp, 2022-05-02:
Hello, same issu. Here is my system info bundle, i hope it will help you.

And thanks for this pluging 
[system info bundle](https://github.com/j7126/OctoPrint-Dashboard/files/8603750/octoprint-systeminfo-20220502165655.zip)
!

mmikem, 2022-05-03:
Hi Jeffery, I noticed that the crazy layer percentages only occur upon first print of a gcode file. Subsequent prints of the same file show the correct layer percentage. Even if printer is power cycled...hope that helps.

mmikem, 2022-05-14:
Hi Jeffery, I have proven my previous troubleshooting results.  This repeatable happens only upon first print of the gcode file.  All subsequent prints never show the crazy layer percentage.  Curious if you have had time to look at this issue. Love the dashboard plug-in, but would love a fix to the layer issue. Thanks.

j7126, 2022-05-15:
Hi, can you please send the gcode file. 
Can you also try installing the development version of dashboard using the plugin manager using this url: `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip` and let me know what happens.

mmikem, 2022-06-10:
Attached is a gcode file that exhibits the issue.  Have not tried the development version yet...

[Box-100x125x45.zip](https://github.com/j7126/OctoPrint-Dashboard/files/8882659/Box-100x125x45.zip)
.

NegussugeN, 2023-06-15:
Chiming in to say same issue here.

[octoprint-systeminfo-20230615110745.zip](https://github.com/j7126/OctoPrint-Dashboard/files/11761226/octoprint-systeminfo-20230615110745.zip)

mmikem, 2023-06-16:
I have found this only happens upon first print of any file. If you cancel the print and start it over the issue goes away for that STL

sasillick, 2023-06-18:
Seeing the same issue - the current layer number does not seem to be updating either.  System info attached
[octoprint-systeminfo-20230618181651.zip](https://github.com/j7126/OctoPrint-Dashboard/files/11782189/octoprint-systeminfo-20230618181651.zip)

barryg41, 2023-06-22:
the beta zip file fixed my layer percentages

sasillick, 2023-06-28:
barryg41 - the Beta version fixed it for me also - thanks