j7126/OctoPrint-Dashboard #340: Manual command

bereldhuin, 2022-04-19:
Hi,

It would be nice to enable manual command widget instead of automated every x seconds.

Here is a simple use case : I have a lamp connected to a Tasmota plug which is used to show what's going on with the webcam, as the room is quite dark. 

I can simply switch it on or off with the following call and get status : 

`curl -s http://192.168.1.46/cm?cmnd=Power%20Toggle | jq -r ".POWER"`

This works fine with dashboard command widget, but i would like to be able to activate it on demand, when clicking on the widget icon, for example.

StefanCohen, 2022-04-21:
FYI: It should be fairly easy to use the [PSUControl Plugin](https://github.com/kantlivelong/OctoPrint-PSUControl) to do the switching and sensing of your lamp as it can use system commands for both. I use that to turn the printer on/off and display the status over an IKEA smart plug.

bereldhuin, 2022-04-22:
Hi,

Thanks for your reply.

I actually use the Tasmota plugin which adds a toogle button in the header bar,  but when Dashboard is in full screen mode, this button is no more visible.