j7126/OctoPrint-Dashboard #342: [Bug]: Old thumbnail shown

giano574, 2022-05-18:
### The problem

I haven't had time to figure out exactly how to reproduce it but my steps were:
1. Upload and print a model through OctoPrint for Cura plugin. Confirm that the thumbnail is shown in the dashboard.
2. Upload a new model and start printing by using OctoPrint for Cura plugin.
3. Confirm that the dashboard has updated, but is showing the old thumbnail.

### Version of OctoPrint

1.7.3

### Platform

Raspberry Pi 3B+

### Operating system running OctoPrint

OctoPi

### Dashboard version

1.19.8

### Browser and version of browser, operating system running browser

Google Chrome

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[octoprint-systeminfo-20220518092850.zip](https://github.com/j7126/OctoPrint-Dashboard/files/8714170/octoprint-systeminfo-20220518092850.zip)

j7126, 2022-05-27:
Can you please check if the thumbnail is shown correctly or not when you view it using the button in the files menu? If it is not shown correctly there, can you please try and see if it is shown correctly there when Dashboard plugin is disabled?

sarf2k4, 2023-02-13:
Hi, I can confirm that this is still a bug that's been around for a while, I am on octoprint 1.8.6 and dashboard 1.19.9

The print thumbnail of this dashboard plugin will show the previously printed/loaded gcode with image on it.

e.g 
gcode 1 has an image of 2 cubes
gcode 2 has an image of 4 pyramids

printing gcode 1 will not show any of the images
printing gcode 2 after gcode 1 will show image of 2 cubes that's from gcode 1 

In some 'perhaps rare' cases, like what I did to check if the layout were correct or not, I pressed the "preview" button multiple times, then reloaded the page via browser reload, the thumbnail is now gone. "slicer thumbnail" plugin works perfectly, maybe you want to try 'pull' the thumbnail data from this plugin?

Perhaps the logic of this dashboard plugin's thumbnail is that, it will be updated once a new gcode is loaded, not during reading the gcode like most should be (I guess)

emis-cz, 2023-11-23:
Hi, I have the same Problem.

Here is a screenshot:
[image]

emis-cz, 2023-12-05:
Hi, it happen if I click on printing direct in the list left.