j7126/OctoPrint-Dashboard #343: [Bug]: Extruded Filament not being reflected unless dashboard refreshed.

Gozania, 2022-05-23:
### The problem

1. As title suggests, filament not being reflected (is being counted) during a print.  The only way to see the current extruded filament is to either A) Refresh ocotoprint window. B) Go fullscreen in dashboard (faster than refreshing octoprint instance) C) Go from fullscreen to normal sized dashboard window.  It appears anytime the dashboard is refreshed it will show the current amount of filament extruded.  Perhaps have this widget "refresh" every 5 seconds or upon every layer completion?

### Version of OctoPrint

1.8.0

### Platform

Raspbery Pi 4 8GB

### Operating system running OctoPrint

Octopi

### Dashboard version

1.19.8

### Python version

3

### Printer model & used firmware incl. version

Ender 3V2 : Jeyers V2.0.2h

### Browser and version of browser, operating system running browser

Firefox 100.0.2 (64 Bit)

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[octoprint-systeminfo-20220522204952.zip](https://github.com/j7126/OctoPrint-Dashboard/files/8751333/octoprint-systeminfo-20220522204952.zip)

j7126, 2022-05-27:
This looks similar to issue #336 which already has a fix ready for the next release. 

You can try it out by installing the version from development branch using this url `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

Please let me know does this solve the problem?

Gozania, 2022-05-28:
> This looks similar to issue #336 which already has a fix ready for the next release.
> 
> You can try it out by installing the version from development branch using this url `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`
> 
> Please let me know does this solve the problem?

Did not solve the problem for me.  Only thing I noticed is the countdown timer has become laggy.  However extruded filament distance still remains the same after almost an hour of time elapsed.

Start of monitoring
[image]

End of monitoring
[image]

As you can see the extruded filament is the same...

Gozania, 2022-06-17:
For whatever reason it works now...

Gozania, 2022-06-17:
Had to reopen this issue...  It appears it worked correctly after a full shutdown of the pi (power included).  The extruded amount will update and function as it should for the first print after reboot, however every subsequent print the extruded amount stays the same regardless of time elapsed during print.

Having just discovered this, I will attempt to repeat the same result by shutting down the pi fully and starting a print after reboot.  Will post findings here.

ochm, 2022-11-05:
Yes, the amount of filament is not updated until page refresh (F5). Trying in Chrome, Firefox ESR, both in latest versions.

j7126, 2022-11-05:
Hi @Gozania @ochm I have just added a new fix to the development branch.

Please reinstall from the development branch and let me know if it is working.
`https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`
(Please note this branch may be unstable)

Gozania, 2022-11-06:
> Hi @Gozania @ochm I have added a new fix to the development branch.
> 
> Please reinstall from the development branch and let me know if it is working. `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip` (Please note this branch may be unstable)

Looking good so far.... Will report after I start my next print.

Gozania, 2022-11-06:
Looks to be working properly now.

Gozania, 2022-11-11:
I spoke too soon... The extruded filament has been fixed, it now displays in real time and dont need to refresh between prints.  However I have been doing some 2 color prints and the time until filament change is not working...  Have done several prints & none of them displayed the time.

j7126, 2022-11-12:
> I spoke too soon... The extruded filament has been fixed, it now displays in real time and dont need to refresh between prints. However I have been doing some 2 color prints and the time until filament change is not working... Have done several prints & none of them displayed the time.

@Gozania Please send the gcode file

Gozania, 2022-11-15:
Here is the Gcode of one of the prints I did.  Others were same stl, just different layers at color change.  This is the latest that did not reflect a filament change time.
[Octopus_spiral_sup_v6_4h35m_$1.20266_22.2798g_7.47005m_210C_PLA.zip](https://github.com/j7126/OctoPrint-Dashboard/files/10015027/Octopus_spiral_sup_v6_4h35m_.1.20266_22.2798g_7.47005m_210C_PLA.zip)