j7126/OctoPrint-Dashboard #346: Translation to German

timhaller, 2022-06-06:
Hello,
I would like to add a translation to German if you are interested

j7126, 2022-07-03:
Hi @timhaller, that would be great.

You can see this issue #284 to se how the french translation was done in the past.

You will need to download the repo and create the file `translations/de/LC_MESSAGES/messages.po`. You can copy this file from the existing french translation and simply change each value of `msgstr` into German in order to add your translation. 

The file contains entries like the following, where `msgid` has the english text and `msgstr` has the translation.
```
#: octoprint_dashboard/static/js/dashboard.js:424
msgid "FullScreen & FullBrowser Mode Buttons"
msgstr "Boutons de mode Plein Écran et Navigateur Plein Écran"
```

I think there may be some parts of the plugin that don't have entries in the file at the moment, however let me know how you go and we can add any missing parts.

I think that should be all that is required for the translation process, however let me know if you encounter any problems.

eschreiner, 2023-03-21:
Hi @j7126 ,

are you still interested in translations to German? I would like to start from the French translations as suggested above...

timhaller, 2023-03-21:
I kind of forgot to be honest but I can do them or assist you with testing if you want

eschreiner, 2023-03-22:
Hi @timhaller I will prepare a ZIP for installation and testing for you until Thursday evening and come back to you then...

eschreiner, 2023-03-23:
Hi @timhaller you can install a preliminary version of the translated dashboard from this URL:

https://github.com/eschreiner/OctoPrint-Dashboard/archive/refs/tags/1.19.10-RC1.zip

Please let me know what you think of it. Suggestions / critique always welcome :-)

Once we are finished with the wording I will create a pull request for @j7126