j7126/OctoPrint-Dashboard #347: [Bug]: Next Filament Change value doesn't make sense

luizbgomide, 2022-06-08:
### The problem

The value displayed in Next Filament Change doesn't make sense, I printed a lot of files with multiple filament changes and in all cases the value was either too short, or too long. I'm not even sure if it is showing time TO change or time OF change, it is very wrong in either case.

By comparison, the plugin TimeToFilament shows the values correctly and accurately (it uses PrintTimeGenius).

### Version of OctoPrint

1.8.1

### Platform

Raspberry Pi 3b

### Operating system running OctoPrint

Octopi 0.18.0

### Dashboard version

1.19.8

### Printer model & used firmware incl. version

Not applicable 

### Browser and version of browser, operating system running browser

Chrome lastest version

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[octoprint-systeminfo-20220607222355.zip](https://github.com/j7126/OctoPrint-Dashboard/files/8857527/octoprint-systeminfo-20220607222355.zip)

Here is a screenshot 23:52 doesn't mean anything here. The next change will happen at around 05:00 and that is something like six and half hours from now.
[image]

luizbgomide, 2022-07-10:
Here is a Gcode that exhibit the behavior
[Corporate Stronghold (Silver Brown Silver) zhop.zip](https://github.com/j7126/OctoPrint-Dashboard/files/9079741/Corporate.Stronghold.Silver.Brown.Silver.zhop.zip)
.

puterboy, 2022-08-26:
I have found the same problem. My next filament change is in about 3 hours (which TimeToFilament gets right) but dashboard shows 20:58 (as in 20hrs, 58minutes).

Seems very broken...

j7126, 2022-10-30:
I will have a look at how TimeToFilament does it

j7126, 2022-10-30:
I have created a fix, it will be included in the next release.
If you want to try it out before then, you can install the development branch using this url
`https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`
Please note that this branch may be unstable.

closed 2022-10-30 by j7126

luizbgomide, 2022-10-31:
Thanks, if I install the development branch will it auto-update when a new version is released?

j7126, 2022-10-31:
Yes