j7126/OctoPrint-Dashboard #349: Long file names exceed Dashboard

CWempe, 2022-06-11:
I am using long file names for my projects and the axceed the dashboard and overlap to the right.

CWempe, 2022-06-11:
I changed some setting of the dashboard for the first time and now it looks good.

Maybe the previous settings were old. 🤷‍♂️ 

[image]

CWempe, 2022-06-11:
Sorry, but I need to reopen this issue again.

When starting a new print job the issue still exists.

[image]

But after pressing F5 to refresh the browser window it is fixed.

j7126, 2022-07-02:
I can't seem to reproduce the issue, can you please share the exact sequence of steps you took before the issue occurred. Can you also please share you octoprint system info bundle.

CWempe, 2022-07-05:
## Steps to reproduce

* open OctoPrint with Chrome
* Power On printer via Tasmota-MQTT Plugin
* Open last project (`.gcode`)
* See long file name

[image]

## octoprint system info bundle

[octoprint-systeminfo-20220705220354.zip](https://github.com/j7126/OctoPrint-Dashboard/files/9049241/octoprint-systeminfo-20220705220354.zip)

commit 833ec885 (2022-08-15): fix #349
closed 2022-08-16 by j7126