j7126/OctoPrint-Dashboard #373: ReOrder Widgets

Sleeepy2, 2023-02-01:
Hi I would l just like the ability to reorder the widgets.
Currently it looks like the order is fixed, with System Info on the top etc.
I would like to be able to reorder them so that I can have webcam at the top then the duration/ layers etc.

csanner, 2023-02-10:
Came looking to request this exact feature. If I have time I'll take a look at the code and see if I can throw something together

Jeppedy, 2023-04-13:
Ditto

SoLongSidekick, 2024-04-09:
> Came looking to request this exact feature. If I have time I'll take a look at the code and see if I can throw something together

With this one comment you brought the hopes of us all under your care. Our dreams reignited with new passion for the possibilities to come, the breath of life we hopelessly sought seemed to no longer be an impossibility. But despite best intentions you have failed us. How cruel to fan the ember of hope, knowing it would never take to flame. What is there to push on for?

(hopefully this is obviously a joke, but who knows maybe the creator of the mod will see this ridiculous comment and be driven to add this feature)

j7126, 2024-04-10:
> With this one comment you brought the hopes of us all under your care. Our dreams reignited with new passion for the possibilities to come, the breath of life we hopelessly sought seemed to no [...]

An attempt at this was made a couple of years ago, however I don't think it went anywhere. I may try to take another attempt at it whenever I have some time. No guarantees as to when that will be though.