j7126/OctoPrint-Dashboard #378: Next Filament change - incorrect for the next change

AlexVercammen, 2023-03-18:
First of all, I love your plug-in. But I've noticed the next filament changes ETA seems incorrect and I would imagine for any subsequent changes.  I have looked at the python code but I don't know how to look into the locally saved data captured by the code to understand what is happening. 
_If you tell me how to look for it, I would gladly investigate the issue and be more helpful._ I am just an occasional programmer nowadays and I have not invested a lot of time into learning OctoPrint plug-in management.
I have included the file I found the issue with.
[OctaTwist_Octagonal-3C_PLA_06mm.zip](https://github.com/j7126/OctoPrint-Dashboard/files/11008827/OctaTwist_Octagonal-3C_PLA_06mm.zip)

tkunchick, 2023-06-26:
I have noticed this is expressed as hours and minutes until next change. If you have "Use 12 hour time for ETA" selected in options, it displays it as a time, rather than a duration.

Would suggest changing so it is always expressed as a duration remaining, or some how do the fancy math to show a time expected for change.

i am not a programmer, no idea how to do either.

AlexVercammen, 2023-06-27:
Cool, I'll try setup to a 12hour ETA since I would prefer a estimated time. 
The issue is when you have more than 1 filament change as the 1st change is finished, the next ETA (duration) is wrong. 
I can see the program parsing the file for the Filament Change code. But I don't understand how to locate the temporary file to validate the parsed values which should be an array like this [(1, 00:01:00), (2, 00:05:00), (3, 00:14:30).
Then, I need to check the code and make sure it is refreshing the displayed ETA using the correct pair from the array.
I have not worked on this since March, my initial posting.

j7126, 2023-07-25:
Hi, this may be related to #347 which should be fixed in the latest 1.19.10 release. Can you please let me know if this issue is still present in this version?