j7126/OctoPrint-Dashboard #384: Total Layer Height Inacurate

jvRobotMan, 2023-04-18:
The total layer height seems to be stuck on the same value print after print, no matter what the height is. Current layer height works but the total does not. See attached image.
[image]

j7126, 2023-07-25:
Please upload the gcode file you are printing when seeing this, and also your octopripnt [system info bundle](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) would be helpful to debug the issue.

mattman86, 2023-08-15:
All of my prints sit at -0.2mmfor the entire print.

mattman86, 2023-08-15:
[info.zip](https://github.com/j7126/OctoPrint-Dashboard/files/12353231/info.zip)