j7126/OctoPrint-Dashboard #388: Add option to disable analysis of uploads

zwimer, 2023-06-18:
## The Problem

When uploading large files, 
The dashboard plugin runs some analysis when uploading files, when these files are large this can be very slow. This entirely freezes/hangs octoprint for minutes on end. A log snippet is attached as an example:
```
octoprint  | 2023-06-18 17:32:23,576 - octoprint.plugins.dashboard - INFO - GcodePreProcessor started processing.
octoprint  | 2023-06-18 17:40:31,203 - octoprint.plugins.dashboard - INFO - GcodePreProcessor found layers: 993
octoprint  | 2023-06-18 17:40:31,230 - octoprint.plugins.dashboard - INFO - GcodePreProcessor found filament changes: 0
octoprint  | 2023-06-18 17:40:31,231 - octoprint.plugins.dashboard - INFO - GcodePreProcessor saving layer count in file metadata
```
Notice the _8 minute_ gap between the analysis starting and finishing. During this time, octoprint is otherwise frozen. This was for an 85 MB file on a raspberry pi 3b.

## The Solution
Provide a way to disable analysis for files on upload.

## Alternatives
I keep this plugin disabled sometimes, for uploads even larger than the 85 MB example above, this can hang octoprint for a very long time.

thijstriemstra, 2023-07-25:
> Add option to disable

if it's the default, i'd call it a bug.

j7126, 2023-07-25:
The analysis is required for many of the features of dashboard to work. I wouldn't call the fact that it runs a bug, however it shouldn't cause octoprint to freeze. 

@zwimer If you send your entire log file I may be able to look into why it freezes octoprint.

I think the best idea in the mean is to add an option to disable it for files over a given size.

zwimer, 2023-07-30:
I won't be able to for a bit, but when my setup is ready again I will post those logs!