j7126/OctoPrint-Dashboard #389: [Bug]: CPU Temperature shows 0C with Octo4a

chis54, 2023-06-22:
### The problem

Running Octo4a on an Android device and the CPU temperature reports as 0C. It does seem to report the operating frequency. 

### Version of OctoPrint

1.9.0

### Platform

Android

### Operating system running OctoPrint

Android 13

### Dashboard version

1.19.9

### Python version

3.10.12

### Printer model & used firmware incl. version

Ender 3 Pro

### Browser and version of browser, operating system running browser

Chrome/Windows 11

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[image]
[octoprint-systeminfo-20230622035423.zip](https://github.com/j7126/OctoPrint-Dashboard/files/11828488/octoprint-systeminfo-20230622035423.zip)

j7126, 2023-07-25:
I am not exactly sure how octo4a works so I can not say this for sure, however it may be the case that it is impossible to read cpu temperature and frequency from the android system.

j7126, 2024-03-10:
I have been investigating a bit more recently into how octo4a works, and I think it actually may be possible after all