j7126/OctoPrint-Dashboard #390: [Bug]: Layer not displayed

TMTYD, 2023-08-02:
### The problem

The layers are not displayed in the dashboard, it says 0/0 layers

### Version of OctoPrint

1.92

### Platform

Raspberry Pi 4

### Operating system running OctoPrint

Debian Bullseye

### Dashboard version

1.19.10

### Python version

3.9.2

### Printer model & used firmware incl. version

Anycubic Kobra Go FW: 1.3.4

### Browser and version of browser, operating system running browser

Firefox, Win 10

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[octoprint-systeminfo-20230803004933.zip](https://github.com/j7126/OctoPrint-Dashboard/files/12245325/octoprint-systeminfo-20230803004933.zip)
[image]

antonyburden, 2024-03-07:
I have the same issue, everything on latest releases - was this ever addressed?

j7126, 2024-03-08:
> I have the same issue, everything on latest releases - was this ever addressed?

@antonyburden this is probably due to a layer analysis issue, which can be caused by a number of factors.
Can you let me know which slicer you are using, and also upload an effected GCODE file?

antonyburden, 2024-03-08:
Thanks for getting back to me - 

I've attached a sample - Using PrusaSlicer 2.7.2

[Joint_0.20mm_220C_PLA.zip](https://github.com/j7126/OctoPrint-Dashboard/files/14541957/Joint_0.20mm_220C_PLA.zip)

j7126, 2024-03-08:
> Thanks for getting back to me -
> 
> I've attached a sample - Using PrusaSlicer 2.7.2
> 
> [Joint_0.20mm_220C_PLA.zip](https://github.com/j7126/OctoPrint-Dashboard/files/14541957/Joint_0.20mm_220C_PLA.zip)

@antonyburden 

It looks like your prusaslicer configuration has been changed from the default. 
Specifically  `printer settings -> expert -> before layer change g-code`
By default it has this:
```gcode
;BEFORE_LAYER_CHANGE
G92 E0
;{layer_z}
```
whereas in your config it is empty.

The dashboard uses the `;BEFORE_LAYER_CHANGE` to detect the layer change, so you at least need that line in there for the dashboard layer detection to work.

antonyburden, 2024-03-09:
Thanks, I now have layer count but no duration!

[image]

j7126, 2024-03-10:
uuh, turns out I somehow missed the original poster of this issue. @TMTYD if you are still having the issue, i will look at it if you upload a gcode file.

closed 2024-03-10 by j7126