j7126/OctoPrint-Dashboard #399: [Bug]: Scaling of bed temperature graph is weird

beelsebob, 2024-02-18:
### The problem

The scaling of the bed temperature graph is weird for my Prusa Mk3.  Here's an example while warming up to print PC-CF:

[image]

As you can see, it seems to be showing the range 100ºC - 1000ºC or something like that, rather than what I'd expect - something like 20ºC to 150ºC.

Doesn't occur when viewing in Firefox, so probably an issue with JS/Safari.

[octoprint-systeminfo-20240218120259.zip](https://github.com/j7126/OctoPrint-Dashboard/files/14322376/octoprint-systeminfo-20240218120259.zip)

### Version of OctoPrint

1.9.3

### Platform

Raspberry Pi 4

### Operating system running OctoPrint

OctoPi

### Dashboard version

1.19.10

### Python version

3.9.2

### Printer model & used firmware incl. version

Prusa i3 Mk3s+, 3.1.12

### Browser and version of browser, operating system running browser

macOS 14.2, Safari 17.2

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

j7126, 2024-03-08:
Yeah most likely an issue with Safari, it can be quite strange at times.

j7126, 2024-03-08:
@beelsebob can you check `Settings -> Dashbaord -> Temperature Gauges -> Bed Max Temp`?

beelsebob, 2024-03-08:
Ahhh, you know I recon that's the issue - it appears that the default max is 100º, which is no good if you're going to start printing PC.  That indication looks an awful lot like it's trying to indicate 4º.  Perhaps there's a `clamp` missing on the temperature somewhere, and a check to show some kind of warning if the temperature is off the scale?

j7126, 2024-03-08:
Done

> I have created a fix, it will be included in the next release!
> If you want to try it out before then, you can install the development branch using this url:
> `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`
> Please note that this branch may be unstable.

closed 2024-03-08 by j7126

beelsebob, 2024-03-08:
Thanks a lot!