j7126/OctoPrint-Dashboard #400: [Bug]: Wrong display of widgets if not Admin account

DerMouse, 2024-03-05:
### The problem

When connecting with an Octoprint Admin access account, everything is displayed properly (below, that's when the printer is connected).
[image]

But when connected with an non-admin account (here, from Operators group), **all widgets** are always displayed : 

[image]
[image]

### Version of OctoPrint

1.9.3

### Platform

Raspberry Pi 3B+

### Operating system running OctoPrint

Gentoo Linux

### Dashboard version

1.19.10

### Python version

3.11

### Printer model & used firmware incl. version

Prusa MK4

### Browser and version of browser, operating system running browser

Firefox 123.0 on Gentoo Linux

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

j7126, 2024-03-08:
Hi, can you please upload the octoprint system info bundle
If you could also can you open the browser console and see if there are any errors there?

DerMouse, 2024-03-08:
I did try (when opening the bug report and a few seconds ago) to upload my system bundle - without success.

Here is the browser console when using a « failing » account.
[image]

j7126, 2024-03-10:
> I did try (when opening the bug report and a few seconds ago) to upload my system bundle - without success.
> 
> Here is the browser console when using a « failing » account.

@DerMouse 

hmm, not sure why you can't upload the system info bundle here, that's strange. Without that I don't really have much info to go on.

Could you try to upload it to some cloud storage (google drive or dropbox or something) and send a link?

DerMouse, 2024-03-13:
> hmm, not sure why you can't upload the system info bundle here, that's strange. Without that I don't really have much info to go on.

Found it. A browser-side plugin doing its stuff.

Here comes the bundle.

[octoprint-systeminfo-20240308235816.zip](https://github.com/j7126/OctoPrint-Dashboard/files/14583009/octoprint-systeminfo-20240308235816.zip)

larp-welt, 2024-04-09:
I have the same problem since the update to OctoPrint 1.10.0rc4.

And here my system bundle.

[octoprint-systeminfo-20240409191718.zip](https://github.com/j7126/OctoPrint-Dashboard/files/14921921/octoprint-systeminfo-20240409191718.zip)

j7126, 2024-04-10:
@DerMouse as for the issue with non-admin accounts, as of yet I have been unable to replicate the issue on my setup.

Can you share if there is any other unusual things about your setup?

Also it appears that in the logs, there is only a very short log history, and you have not logged in with a non-admin account since the logs were cleared, as only admin logins are present in the logs. Can you sign in with the non-admin account which experiences the issue and then send the system bundle?

larp-welt, 2024-04-11:
@j7126 that's true, the new version fixed the problem for. 

Thank you very much! Octoprint without the Dashboard is not the same thing! :+1:

DerMouse, 2024-04-12:
Hello,

> @DerMouse as for the issue with non-admin accounts, as of yet I have been unable to replicate the issue on my setup.
> 
> Can you share if there is any other unusual things about your setup?

Since we had similar problems with several other plugins, we got somehow suspicious. We made a brand-new, clean, from scratch install of octoprint (we are installing everything on a Gentoo Linux Raspi, not using OctoPi), and of course all the problems disappeared as by magic.

So our previous installation (from an Octoprint « backup and restore ») must have had something weird causing the misbehaviours.

Sorry for wasting your time. 
Closing this issue, since for the time being (and hopefully the future) the plugin is working flawlessly.

DerMouse, 2024-04-12:
I said... closing this issue.

closed 2024-04-12 by DerMouse