j7126/OctoPrint-Dashboard #403: Not working on latest octoprint 1.10.0rc4

tigercjn, 2024-04-09:
### The problem

Just headings shows without data.
[image]

### Version of OctoPrint

 1.10.0rc4

### Platform

Pi 3b+

### Operating system running OctoPrint

Octopi

### Dashboard version

1.19.10

### Printer model & used firmware incl. version

snapmaker 2.0

### Browser and version of browser, operating system running browser

Chrome Version 123.0.6312.106 (Official Build) (64-bit)

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[image]
[octoprint-systeminfo-20240409120222.zip](https://github.com/j7126/OctoPrint-Dashboard/files/14912532/octoprint-systeminfo-20240409120222.zip)

jneilliii, 2024-04-09:
To add to this issue, it appears to be related to the printer profile binding.

[image]

j7126, 2024-04-09:
Yeah it appears that currentProfileData() is undefined for some reason

j7126, 2024-04-09:
fixed in latest release 1.19.11

closed 2024-04-09 by j7126